facebookresearch/pyrobot #3: robot name parsing from config file is incomplete

harryrobotics, 2019-06-22:
I created config file name : kobuki_config or husky_config for kobuki and husky robot.
When I run : 

robot = Robot(‘kobuki’) or 
robot = Robot(‘husky’)

Program will throw an error that : it can’t find the robot name husk or kobuk 
The letter i and y are not capture when parsing the file name robot_config.py .

kalyanvasudev, 2019-06-22:
Thank you for bringing this to our attention. The bug has been resolved in the commit a27649d6751c8e15111d3e391725419af41ccd16

Please let us know if have any more questions or doubts on how to add a new robot to PyRobot.
We are excited to see more robots on this platform :)

sushil-bharati, 2019-06-22:
nice catch, @kalyanvasudev

closed 2019-06-22 by kalyanvasudev