facebookresearch/pyrobot #5: Documentation to control different motors

mkirank, 2019-06-23:
Is there a way to control different motor's like DC, stepper and servo. I could not find this in the documentation.

harryrobotics, 2019-06-23:
I think hardware controller should be implemented in the ROS layer. PyRobot only pass the velocity command to ROS layer. ROS will take care of writing to hardware.

kalyanvasudev, 2019-06-23:
If new robot hardware needs to be interfaced with our framework, as mentioned in the documentation, the person adding the robot would have to write the ROS mid-level and low-level layers. We would be happy to help if you get stuck in this process :) We will also be adding more robots and simulators to PyRobot soon!

The motivation is that having implemented all the ROS-layers, the low-level code and interfaced with PyRobot, PyRobot makes it easy for people to work with robots. Especially, newcomers to robotics and people who have never worked ROS.

If a robot is already supported by PyRobot, newcomers wouldn't have to worry about the daunting low-level (drivers, controls, sensing, etc) and mid-level software (planning, collision checking, kinematics, etc). This should make it easy for them to do their cool ML and CV research :)

As the low-level and mid-level stack need to be only written once, PyRobot helps avoid duplicated and wasted effort.

Here is a useful link - https://ai.facebook.com/blog/open-sourcing-pyrobot-to-accelerate-ai-robotics-research/

closed 2019-06-23 by kalyanvasudev