facebookresearch/pyrobot #7: Installation script: Can't run catkin_init_workspace in ~/camera_ws folder

harryrobotics, 2019-06-24:
I ran this command:
```
chmod +x locobot_install_all.sh 
./locobot_install_all.sh
```
Until running cakin_init_workspace in ~/camera_ws folder, I got the error like this:
```
harry@harry:~/camera_ws/src$ catkin_init_workspace Could neither symlink nor copy file "/opt/ros/kinetic/share/catkin/cmake/toplevel.cmake" to "/home/harry/camera_ws/src/CMakeLists.txt": - [Errno 13] Permission denied - [Errno 13] Permission denied: '/home/harry/camera_ws/src/CMakeLists.txt
```
Later I found out that if I run this script with root permission , the the `camera_ws` folder will be created with root permission. And then catkin_init_workspace will not work.
I had to removed the  `camera_ws`` folder, recreate it with user permission only, then I could run ```catkin_init_workspace```.

The lesson learnt that don't create the workspace folder with root permission.
However, if I don't run script with sudo , I will be stopped at : "Installing camera dependencies..."

Does it only happen to me? Please help to check if you can reproduce the issue. Thank you.

kalyanvasudev, 2019-06-24:
Hi Harry, I'm having a hard time reproducing the error you mentioned on my machines. Everything seems to be working fine.  

Do you have admin rights for the account from which you are running the installation script? 

Any other additional info you can provide to help with debugging would be appreciated.

harry-savioke, 2019-06-26:
Sorry about the false alarm. I would have messed up somewhere with my file system. Should I run this scripts on my computer? I ran it and my laptop webcam stops working. When I remove the librealsense , the camera works again.

harry-savioke, 2019-06-26:
I saw your recent commit, if only use simulation, I don't need to install librealsense package. Correct me if I am wrong.

kalyanvasudev, 2019-06-26:
If you are using only gazebo simulation, you don't need to install realsense related packages. However, the current installation script is designed to install everything. I made a modular installation script in this pull request - https://github.com/facebookresearch/pyrobot/pull/9

If you wish to use this new script, run with the command "./locobot_install_all.sh sim"

I will merge this new installation script with the main repo after fully testing it on multiple machines in a day or two.

closed 2019-06-29 by harryrobotics