facebookresearch/pyrobot #11: _loop_angle_pub_cmd crash because no velocity info is given

Pandhariix, 2019-06-27:
If I implement the arm of a new robot but can't get any velocity / effort data for the arm's joints (through a JointState message), the __go_home__ method will fail.

This is due to the fact that __go_home__ calls ___loop_angle_pub_cmd__, in which the following snippet can be found: 
```python
vels = self.get_joint_velocities()
es_time = time.time() - start_time

if es_time > 0.5 and np.max(np.abs(vels)) < 0.01:
    vel_zero_times += 1
```

The if instruction will fail with a value error, the vels list being empty. 

My question is the following, would it be more pertinent to override the ___loop_angle_pub_cmd__ in the robot's Arm class, or to modify the original one in the core file ?

kalyanvasudev, 2019-06-27:
Hi Maxime, which robot are you trying to add? I would go about overriding **_loop_angle_pub_cmd** function in your robot's Arm class implementation.

Pandhariix, 2019-06-27:
Hi Kalyan, thank you for your quick answer. I was thinking about adding the Pepper robot to the framework.

kalyanvasudev, 2019-06-27:
looking forward to see Pepper on our platform :) 
Please let me know if you have anymore questions. Thanks!

closed 2019-06-27 by kalyanvasudev