facebookresearch/pyrobot #18: Cannot import PyKDL

hzyjerry, 2019-07-07:
Hi, I am using pyrobot on Ubuntu 16.04.

After successful installation, I executed `python base_position_control.py` and got the following error:

```
Traceback (most recent call last):
  File "base_position_control.py", line 12, in <module>
    from pyrobot import Robot
  File "/home/jerrhe/miniconda3/envs/ros/lib/python2.7/site-packages/pyrobot/__init__.py", line 6, in <module>
    from pyrobot.core import Robot
  File "/home/jerrhe/miniconda3/envs/ros/lib/python2.7/site-packages/pyrobot/core.py", line 16, in <module>
    import PyKDL as kdl
ImportError: No module named sip
```

I also tried `sudo apt-get install python-sip` but doesn't seem to help. Have you any suggestion for the issue? Thanks!

kalyanvasudev, 2019-07-07:
Try removing SIP and installing it using "pip install sip"

kalyanvasudev, 2019-07-07:
Also, PyRobot python api is configured to work with the virtual environment setup we provided and Python2.7  

Looks like you are using a different environment?

kalyanvasudev, 2019-09-03:
I assume you were to able the resolve the issue.. Closing this issue for now.

closed 2019-09-03 by kalyanvasudev