facebookresearch/pyrobot #20: Using PyRobot to control a custom robot

simplyglobal, 2019-07-19:
I am building a robot that can move around my front yard, identify certain objects ("gum tree seed pods") and pick them up. I have started [documenting my journey](https://medium.com/@manishdharod/using-deep-learning-to-build-a-robo-yard-cleaner-4a675db26cf6) as I go along. I am hoping to use it to control the robot arm. I looked into your documentation and had two questions:

1. If I want to start getting familiar with PyRobot APIs and just control one or two motors (i.e. 1-2 DOF robot arm) through an Arduino Uno board, do you have pointers on examples or tutorials to do this? I looked at the Locobot examples but found it a little harder to understand how to get this to work for a custom robot arm and scaling it from 1 to 5 DOF one step at a time if needed?

2. Also, based on my understanding Locobot and Sawyer have their own controllers and PyRobot is able to work with those controllers. In my case, since its Arduino, wondering what part of the code do I need to modify to get it to work?

Thanks
Manish

kalyanvasudev, 2019-07-19:
Hi,

1. I encourage you to look at our website - pyrobot.org
Here, we have instructions on how to add a new robot - https://www.pyrobot.org/docs/new_robot_support

2. Our framework is built on ROS. You need a PC capable of running Ubuntu 16.04 with ROS-Kinetic to run our libraries.

harryrobotics, 2019-07-20:
Hi Manish,

PyRobot doesn't interact directly with hardware, therefore, you can't use PyRobot to control Arduino. PyRobot will interact with ROS layer and subsequently, ROS will interact with Arduino. 

I suggest that you should install ROS on your Linux laptop and figure out how ROC interacts with Arduino first. As I know, Arduino has a library supporting the ROS messages. Arduino will act as a node - publish and subscribe the information from one host computer (e.g Your Linux Laptop). In short, just try to use and play with ROS first. When knowing more about ROS and python, you will intuitively figure out how PyRobot works.

kalyanvasudev, 2019-09-03:
Closing this issue as @harryrobotics's reply answers your question.

closed 2019-09-03 by kalyanvasudev