facebookresearch/pyrobot #23: Installation issue with pip

ozbeluga, 2019-07-31:
After following the steps you shared for installation I am having trouble running locobot_install_all.sh. I get the following:

Traceback (most recent call last):
  File "/usr/bin/pip", line 9, in <module>
    from pip import main
ImportError: cannot import name main

I searched online, but only got confused more about the issue. Do you have any idea about this?

Thank you!

kalyanvasudev, 2019-08-01:
Hi,

Could you please provide more information about your setup, etc?

Also, could you post the entire error you saw?

Siddu733, 2019-08-14:
I have not got any issues while installing pyrobot. I have done using "pip install pyrobot" command in python terminal. For any issues, visit https://pypi.org/project/pyrobot/ .

kalyanvasudev, 2019-08-14:
Hi @Siddu733 , the pyrobot you mentioned in the link is a different project. It is not associated with this project or this repository.

kalyanvasudev, 2019-09-03:
Closing this issue as it is unrelated.

closed 2019-09-03 by kalyanvasudev