facebookresearch/pyrobot #24: I follow the "./locobot_install_all.sh" and have problem

Ljh0121, 2019-08-21:
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
locobot_description: No definition of [gazebo] for OS version []
ca_driver: No definition of [libcreate] for OS [debian]
dynamixel_workbench_operators: No definition of [yaml-cpp] for OS version []
ca_tools: No definition of [joy_teleop] for OS [debian]
dynamixel_workbench_controllers: No definition of [eigen] for OS version []
locobot_gazebo: No definition of [gazebo] for OS version []

kalyanvasudev, 2019-08-22:
Hi, Could you please provide more information about your setup for me to help you resolve the issue?

Ljh0121, 2019-08-22:
> Hi, Could you please provide more information about your setup for me to help you resolve the issue?

This problem has been solved, but there is another problem：
make[2]: *** [/home/ethan/low_cost_ws/devel/lib/orb_slam2_ros/orb_slam2_rgbd] Error 1
CMakeFiles/Makefile2:10735: recipe for target 'pyrobot/robots/LoCoBot/locobot_navigation/orb_slam2_ros/CMakeFiles/orb_slam2_rgbd.dir/all' failed
make[1]: *** [pyrobot/robots/LoCoBot/locobot_navigation/orb_slam2_ros/CMakeFiles/orb_slam2_rgbd.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....

I found there is not the  executable file named "orb_slam2_rgbd" .

closed 2019-08-23 by Ljh0121

tkim61, 2021-05-26:
Hi, I am working with pyrobot now - I encountered a similar issue.  To get around the first one I upgraded from ubuntu 16.04 and ROS kinetic (which has reached end of life just recently) to ubuntu 18.04 and ROS melodic.  I am also getting errors with CMake and orb_slam2_ros build files.  I don't see an 'all' directory or anything mentioning all in the files in the system.  It should be noted that those files are in ~/low_cost_ws/**build**/pyrobot..., not src/pyrobot...