facebookresearch/pyrobot #27: Unable to Run SLAM

vikashranjan, 2019-08-29:
I have built a LoCoBot without the robotic arm and movable camera. I get the following issue when I run SLAM.

LoCoBot installation is done using locobot_install_all.sh .

I have commented calibration_tf_broadcaster as suggested in issue #26 in main.launch and updated permission to /dev/ttyUSB0.

Command : roslaunch locobot_control main.launch use_base:=true use_vslam:=true use_camera:=true

Hardware :

    Intel NUC i5 with 8 Gb Ram
    Intel D435 camera
    Base Kobuki base

Error:

========
... logging to /home/vikash/.ros/log/978b8314-cab2-11e9-97a9-00c2c6bcf3fe/roslaunch-vikash-desktop-2101.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://vikash-desktop:46073/

SUMMARY

PARAMETERS
 * /bumper2pointcloud/pointcloud_radius: 0.24
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/clip_distance: -1.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_imu: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/infra1_fps: 30
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_height: 480
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra1_width: 640
 * /camera/realsense2_camera/infra2_fps: 30
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_height: 480
[... 542 lines cut ...]
[ INFO] [1567120403.740436127]: Done Setting Dynamic reconfig parameters.
 30/08 07:13:23,758 WARNING [139723648333568] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.
[ WARN] [1567120403.759075034]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)
 30/08 07:13:23,761 WARNING [139723637466880] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.
Starting context monitors...
Context monitors started.
Loading 'move_group/ApplyPlanningSceneService'...
Loading 'move_group/ClearOctomapService'...
Loading 'move_group/MoveGroupCartesianPathService'...
Loading 'move_group/MoveGroupExecuteTrajectoryAction'...
Loading 'move_group/MoveGroupGetPlanningSceneService'...
Loading 'move_group/MoveGroupKinematicsService'...
Loading 'move_group/MoveGroupMoveAction'...
Loading 'move_group/MoveGroupPickPlaceAction'...
Loading 'move_group/MoveGroupPlanService'...
Loading 'move_group/MoveGroupQueryPlannersService'...
Loading 'move_group/MoveGroupStateValidationService'...

You can start planning now!

[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

kalyanvasudev, 2019-08-30:
Please check if your camera is working properly.  As mentioned the issue https://github.com/IntelRealSense/realsense-ros/issues/503, check if unplugging and re-plugging the realsense solves your issue.

As mentioned on the main page of the repo, realsense should be connected to the computer while running the locobot_install_all.sh 

A few tools to debug if camera is working properly,

1. Run "roslaunch locobot_control main.launch use_base:=true use_camera:=true" and go to pyrobot tests folder and run pytest test_camera.py -v

If the above test return "None" as the image. That means the camera is not properly setup.
To confirm this, check if "rostopic echo /camera/aligned_depth_to_color/image_raw" prints any image messages. If this also fails/timesout, this means that your realsense is not setup properly.

2. If 1 fails, possible solutions to this are,

- Try the solutions mentioned in this related issue - https://github.com/IntelRealSense/realsense-ros/issues/503

- Try updating your realsense drivers and firmware as mentioned in the FAQ here - https://www.pyrobot.org/docs/faq

- Delete everything, connect realsens to your computer via the usb-c cable provide in the intelrealsense box and re-run the installation script.

vikashranjan, 2019-08-31:
Thanks for your reply. I am able to see images published to ```/camera/aligned_depth_to_color/image_raw``` . It seems the issue maybe with the kobuki base. I will try to debug the base independently and update the ticket.

vikashranjan, 2019-09-05:
I updated firmware for both D435 Camera and Base. I also did a fresh install from scratch.SLAM still doesn't work.
Could it be related to ```[DynamixelDriver] Failed to open the port! ``` .My camera is fixed.

kalyanvasudev, 2019-09-05:
If your camera is working fine, based the error dynamixel message presented, it could be that the Pan and Tilt motors on which the camera is mounted are properly not recognized.

What is the out the following test?

Run ```roslaunch locobot_control main.launch use_base:=true use_camera:=true''' and go to pyrobot tests folder and ```run pytest test_camera.py -v'''

vikashranjan, 2019-09-05:
I get the following error after running ```pytest test_camera.py -v```

=========================== 6 failed in 3.72 seconds ===========================
(pyenv_pyrobot) vikash@vikash-desktop:~/low_cost_ws/src/pyrobot/tests$ pytest test_camera.py -v
============================= test session starts ==============================
platform linux2 -- Python 2.7.12, pytest-4.6.5, py-1.8.0, pluggy-0.12.0 -- /home/vikash/pyenv_pyrobot/bin/python2.7
cachedir: .pytest_cache
metadata: {'Python': '2.7.12', 'Platform': 'Linux-4.15.0-60-generic-x86_64-with-Ubuntu-16.04-xenial', 'Packages': {'py': '1.8.0', 'pytest': '4.6.5', 'pluggy': '0.12.0'}, 'Plugins': {u'html': u'1.20.0', u'metadata': u'1.8.0', u'cov': u'2.6.1'}}
rootdir: /home/vikash/low_cost_ws/src/pyrobot
plugins: cov-2.6.1, html-1.20.0, metadata-1.8.0
collected 6 items                                                              

test_camera.py::test_position_control[target_position0] FAILED           [ 16%]
test_camera.py::test_position_control[target_position1] FAILED           [ 33%]
test_camera.py::test_position_control[target_position2] FAILED           [ 50%]
test_camera.py::test_position_control[target_position3] FAILED           [ 66%]
test_camera.py::test_position_control[target_position4] FAILED           [ 83%]
test_camera.py::test_get_images PASSED

kalyanvasudev, 2019-09-06:
Based on the test results it looks like there is some problem with your camera pan and tilt (dynamixel motors).  Please make sure all the cables are properly wired for the motors. Please make sure you also followed all the assembly instructions mention on the LoCoBot assembly website ( locobot.org ) accurately.

On a side note, it looks like the realsense camera is working fine.

vikashranjan, 2019-09-06:
I commented ```dynamixel_controllers.launch``` in main.launch and I can see the ORB_SLAM2 Map Viewer and rviz launch but the map creation doesn't start.The robot stays still.

1. Does the robot move automatically during map creation or needs to manually controlled via kobuki_keyop ?
2.How and where is the created map stored ?
3.Can SLAM be reset and run multiple times ?
4.Is it possible to have multiple maps for the same area ?

vikashranjan, 2019-09-06:
> Based on the test results it looks like there is some problem with your camera pan and tilt (dynamixel motors). Please make sure all the cables are properly wired for the motors. Please make sure [...]

I  am not using dynamixel motors yet . I have a fixed camera.Are they mandatory for slam to run ?

kalyanvasudev, 2019-09-06:
Here are the answers to your questions,
1. No, the robot doesn't move autmatically. You need to move it with teleop to build the map.
2. Here is an example on how to get map points,
https://github.com/facebookresearch/pyrobot/blob/master/examples/locobot/navigation/vis_3d_map.py
3. No, you would have to restart the node. Reset does seem like a handy feature to have. I will look into it and add one soon.
4.  do you mean like different looking maps over different trials? Noise in the map building could lead to slightly different looking maps.

vikashranjan, 2019-09-07:
Thanks for your answers !.
I get the following error when I run ```vis_3d_map.py```

```
[WARN] [1567873738.409640]: Neither use_arm, nor use_sim, is not set to True when the LoCoBot driver is launched.You may not be able to command the arm correctly using PyRobot!!!
Traceback (most recent call last):
  File "vis_3d_map.py", line 55, in <module>
    main()
  File "vis_3d_map.py", line 25, in main
    bot = Robot('locobot')
  File "/home/vikash/pyenv_pyrobot/local/lib/python2.7/site-packages/pyrobot/core.py", line 108, in __init__
    self.base = base_class(self.configs, **base_config)
  File "/home/vikash/pyenv_pyrobot/local/lib/python2.7/site-packages/pyrobot/locobot/base.py", line 231, in __init__
    self.base_state = BaseState(base, self.build_map, map_img_dir, configs)
  File "/home/vikash/pyenv_pyrobot/local/lib/python2.7/site-packages/pyrobot/locobot/base.py", line 152, in __init__
    z_max=self.configs.BASE.Z_MAX_TRESHOLD_OCC_MAP)
  File "/home/vikash/low_cost_ws/src/pyrobot/robots/LoCoBot/locobot_navigation/orb_slam2_ros/scripts/orb_slam2_ros/vslam.py", line 96, in __init__
    self.base_frame)
  File "/home/vikash/low_cost_ws/src/pyrobot/robots/LoCoBot/locobot_navigation/orb_slam2_ros/scripts/orb_slam2_ros/vslam.py", line 354, in get_link_transform
    src)
  File "/home/vikash/low_cost_ws/src/pyrobot/robots/LoCoBot/locobot_navigation/orb_slam2_ros/scripts/orb_slam2_ros/vslam.py", line 391, in _get_tf_transform
    rospy.Duration(3))
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/tf/listener.py", line 76, in waitForTransform
    raise tf2_ros.TransformException(error_msg or "no such transformation: \"%s\" -> \"%s\"" % (source_frame, target_frame))
tf2.TransformException: Could not find a connection between 'base_link' and 'camera_color_optical_frame' because they are not part of the same tree.Tf has two or more unconnected trees.
Exception AttributeError: "'BaseState' object has no attribute 'subscribers'" in <bound method BaseState.__del__ of <pyrobot.locobot.base.BaseState object at 0x7f9922d0fc90>> ignored

```

kalyanvasudev, 2019-09-07:
The error is attributed to your earlier attributed problem with pan and tilt motors. Please refer to my earlier reply...

Based on the test results it looks like there is some problem with your camera pan and tilt (dynamixel motors). Please make sure all the cables are properly wired for the motors. Please make sure you also followed all the assembly instructions mention on the LoCoBot assembly website ( locobot.org ) accurately.

On a side note, it looks like the realsense camera is working fine.

vikashranjan, 2019-09-08:
Is it possible to use a fixed camera without pan and tilt motors ? .I haven't installed dynamixel motors for pan and tilt. Using a fixed camera mounted on kobuki base.

vikashranjan, 2019-09-10:
Thanks for the help!

closed 2019-09-10 by vikashranjan

kalyanvasudev, 2019-09-18:
Hi, I hope your were able to resolve the issue? The easiest way to fix this issue if you don't have access to camera pan and tilt motors is to,
1. Comment out dynamixel related commands to locobot_control/main.launch.
2. Publish fake joint angle values for pan and tilt motors. (This is needed to ensure a transformation exists form the robot base to the camera.