facebookresearch/pyrobot #30: Wrong Dynamixel Motor Power Hub Link on Locobot Page

msr-peng, 2019-09-21:
Motor power hub's link on [Locobot components page](https://locobot-website.netlify.com/docs/k_step_2) leads to the wrong product.
[image]
     
It leads to the [following power hub](https://www.trossenrobotics.com/6-port-ax-mx-power-hub):
[image]
     
However, **the right power hub's port is slightly different**. [Here](https://www.trossenrobotics.com/3-pin-x-series-power-hub.aspx) is the right one:
[image]
     
For those who don't buy the arm which come with the motor power hub, it's very easy to buy the wrong one given the instruction on locobot page. So please update the correct link.

Dhiraj100892, 2019-09-21:
Hi @msr-peng ,

Thanks for pointing it out and I am sorry for the mistake. I have updated it on the website.

msr-peng, 2019-09-21:
> Hi @msr-peng ,
> 
> Thanks for pointing it out and I am sorry for the mistake. I have updated it on the website.

Cool~ Then I'll close this issue.

closed 2019-09-21 by msr-peng