facebookresearch/pyrobot #31: Ubuntu pops up "crash error" during install

iovsiann, 2019-09-27:
## Steps to reproduce

  1. Ubuntu 16.04.6
  2. ./locobot_install_all.sh -t full -p 2
  3. _____

## Observed Results

PyRobot still installs, but Ubuntu reports a crash caused by ros-kinetic-librealsense. ros-kinetic-librealsense is not to be confused with ros-kinetic-librealsense2, which is needed for Intel RealSense D435. ros-kinetic-librealsense seems to be part of unused "junk" dependencies required by ros-kinetic-turtlebot-* and ros-kinetic-rgbd-launch in locobot_install_all.sh. By the way, ros-kinetic-librealsense itself requires even more unused "junk" like linux-headers-4.4.0-164 for uvc patching.

[image]

## Expected Results

Installation without Ubuntu complaining about crash error.

## Relevant Code

Just run install Ubuntu 16.04.6 and run ./locobot_install_all.sh -t full -p 2

swtrossen, 2019-10-17:
I have run into this issue as well. According to [this](https://github.com/IntelRealSense/librealsense/issues/2402), it seems that ros-kinetic-librealsense only installs correctly with specific Linux kernels. In the link, it mentions that it works on Linux kernel 4.4. I've tried several versions of kernel 4.4 but had no luck either because the package failed to install or internet didn't work in that kernel subversion. My guess is that there is a sub-version of kernel 4.4 that works, but I just haven't found it yet. Anyway, I don't think it's a big issue that it fails to properly install since it does not seem to be used in Pyrobot, but I agree that it would be nice to have this fixed.

kalyanvasudev, 2019-12-19:
this issue has been addressed in ros-melodic ubuntu 18.04 branch

closed 2019-12-19 by kalyanvasudev