facebookresearch/pyrobot #32: Issue with active camera on LoCoBot

bhyang, 2019-10-02:
## Steps to reproduce

I built the LoCoBot with active camera, but I can't get the active camera working. Only hardware difference is that I'm running everything on a NVIDIA Jetson instead of a NUC.
1. Run `roslaunch locobot_control main.launch use_camera:=true use_arm:=true`

## Observed Results
Here's what I get when I run the above command
```
(pyenv_pyrobot) nvidia@tegra-ubuntu:~/low_cost_ws/src/pyrobot/robots/LoCoBot/locobot_control/launch$ roslaunch locobot_control main.launch use_arm:=true use_cam
era:=true
... logging to /home/nvidia/.ros/log/b2cd9c3a-dfe2-11e9-a017-00044bc76af9/roslaunch-tegra-ubuntu-7183.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://tegra-ubuntu:44271/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -1.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
[... 382 lines cut ...]
Loading 'move_group/MoveGroupGetPlanningSceneService'...
Loading 'move_group/MoveGroupKinematicsService'...
Loading 'move_group/MoveGroupMoveAction'...
Loading 'move_group/MoveGroupPickPlaceAction'...
Loading 'move_group/MoveGroupPlanService'...
Loading 'move_group/MoveGroupQueryPlannersService'...
Loading 'move_group/MoveGroupStateValidationService'...

You can start planning now!

[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```
The same issue doesn't occur if I run `roslaunch locobot_control main.launch use_arm:=true`. A similar issue occurs if I run `roslaunch locobot_control main.launch use_camera:=true`.

## Expected Results

I expected the node for the active camera to start running so I could run camera calibration.

## Relevant Code

I haven't modified any of the launch scripts or source code. I've checked that the servos are all ID'd properly and have baudrate of 1000000. The fact that the arm control works in isolation tells me that it's an issue with the active camera specifically. From observation, it also seems that the servos aren't blinking red when all the errors pop up so I don't believe the servos are all actually experiencing overload/overheating/etc.

kalyanvasudev, 2019-10-03:
As mentioned in the installation instructions, was the realsense camera connected to the computer (via the appropirate USB3- TypeC ) cable while running the installation script? If not, please make sure you do.

Also, please consider upgrading your realsense camera firmware using the instructions provided int his link - https://www.pyrobot.org/docs/faq [Faq No: 4]

This older issue has a few camera debugging tricks - https://github.com/facebookresearch/pyrobot/issues/27#issuecomment-526697002

bhyang, 2019-10-04:
The camera was connected to the computer during installation. I've updated the camera firmware, but that didn't fix the issue.

To verify that the camera works, I checked the camera output in `realsense-viewer`. I also ran `roslaunch realsense2_camera rs_camera.launch` and `rosrun image_view image_view image:=/camera/color/image_raw`. In both cases, the camera seemed to be streaming properly.

From inspecting the logs, it also seems like the realsense node isn't causing these issues. I think it has something to do with the pan/tilt motors?

I tried operating the arm standalone, and it works fine. I also tried connecting the arm using the long cord originally for motor 8, and it still works, so I don't think there's an issue with the connecting cord.

When I try to use the pan/tilt motors using `roslaunch locobot_control change_id.launch`, I'm not running into any issues either. I've also tried resetting/rebooting the pan/tilt motors, but the issue remains.

Here's the newest log from running `roslaunch locobot_control main.launch use_camera:=true`
```
(pyenv_pyrobot) nvidia@tegra-ubuntu:~/low_cost_ws$ roslaunch locobot_control main.launch use_camera:=true
... logging to /home/nvidia/.ros/log/c17a5de8-dfe4-11e9-82a1-00044bc76af9/roslaunch-tegra-ubuntu-5275.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://tegra-ubuntu:36297/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -1.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
[... 335 lines cut ...]
 25/09 15:35:30,194 WARNING [546450694544] (sensor.cpp:455) Frame received with streaming inactive,Depth0, Arrived,0.000000 1569450930193.958252
terminate called without an active exception
Context monitors started.
Loading 'move_group/ApplyPlanningSceneService'...
Loading 'move_group/ClearOctomapService'...
Loading 'move_group/MoveGroupCartesianPathService'...
Loading 'move_group/MoveGroupExecuteTrajectoryAction'...
Loading 'move_group/MoveGroupGetPlanningSceneService'...
Loading 'move_group/MoveGroupKinematicsService'...
Loading 'move_group/MoveGroupMoveAction'...
Loading 'move_group/MoveGroupPickPlaceAction'...
Loading 'move_group/MoveGroupPlanService'...
Loading 'move_group/MoveGroupQueryPlannersService'...
Loading 'move_group/MoveGroupStateValidationService'...

You can start planning now!

[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

Any help would be appreciated. Thanks!

kalyanvasudev, 2019-10-07:
To verify that the camera works, I checked the camera output in realsense-viewer. I also ran roslaunch realsense2_camera rs_camera.launch and rosrun image_view image_view image:=/camera/color/image_raw. In both cases, the camera seemed to be streaming properly.

Hi, Just to make sure camera is truly not the source of the error, prior to your earlier camera experiment can you do the following and confirm everything is fine?

Run, ```roslaunch realsense2_camera rs_camera.launch''' and then in a new terminal do ```rosrun image_view image_view image:=/camera/depth/image_rect_raw'''

Are you able to see the depth image?

bhyang, 2019-10-07:
Okay with some help I've fixed the issue. I was using [these](https://www.trossenrobotics.com/store/p/6681-3-Pin-Coupler-25-Pack.aspx) since I didn't get the long 3-pin cable from Trossen, but this was responsible for the issue. Not using the extenders seems to resolve the errors. I don't believe the camera played a role here (I could see the depth images).
Thanks!

closed 2019-10-07 by bhyang