facebookresearch/pyrobot #41: Image Quality

roozbehm, 2019-11-08:
The images that we obtain from the camera are a bit blue-ish and they are very different from the actual lighting of the room. We were wondering if there is any fix for this. I have attached an image for reference. That is a white couch. 
[image]

Dhiraj100892, 2019-11-08:
Is it possible to know which USB cable are you guys using for connecting camera to NUC?

Dhiraj100892, 2019-11-08:
If camera does not get enough power, then it won't work properly. Perhaps charging the battery and trying again might fix the problem. Also, can you try plugging the camera into the orange USB port on the NUC which can supply more current?

Also can you tell us what exact command are you guys running?

roozbehm, 2019-11-08:
We used the cable that came with the robot. 
Regarding the command, we use the following code: https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_viewer_example.py#L27

Dhiraj100892, 2019-11-08:
The earlier robot from Trossen where shipped with USB2.0 cable. Is it possible to post the picture of usb A port? I just wanted to make sure you are using 3.0.

swinterbotix, 2019-11-19:
Another way you can check if you are using a USB 2 cable is by typing `rs-enumerate-devices` in your terminal and reading the output. I believe it should specify it under Device info: Usb Type Descriptor

kalyanvasudev, 2019-11-26:
Hi, is the issue resolved by using the right usb 3.0 cable ?

roozbehm, 2019-11-26:
No, it was not resolved. The output of `rs-enumerate-devices` is Usb Type Descriptor 3.2.

dhruvbatra, 2019-11-28:
@roozbehm -- @erikwijmans and @mathfac have experienced this "images appearing blue" problem in past. It was due to RGB vs BGR ordering used by OpenCV. It's a long shot, but might want to check that.

roozbehm, 2019-11-29:
We had tried that as well. It didn’t fix the problem. I think the camera quality is not that good.

kalyanvasudev, 2019-12-20:
can you try testing the realsense with the following commands?
cd /usr/local/bin
$ ./realsense-viewer

kalyanvasudev, 2020-03-05:
Closing this issue due to inactivity. Please re-open this if you have further questions

closed 2020-03-05 by kalyanvasudev