facebookresearch/pyrobot #43: LoCoBot Launch File Crashes Every 15-20 Minutes

jgonik, 2019-11-18:
I am running data collection code on a LoCoBot where the robot randomly takes one of four actions (turn left, turn right, move forward, do nothing) and captures an image with the camera after each action. Approximately every 15-20 minutes, the launch file crashes and must be restarted. This is how I am launching the robot:

```
roslaunch locobot_control main.launch use_base:=true use_camera:=true use_rviz:=false
```

Here is the error message outputted when the launch file crashes:

```
[ERROR] [1574104984.338762683]: [TxRxResult] There is no status packet!
[ERROR] [1574104984.338863115]: groupSyncRead getdata failed
[ERROR] [1574104984.338922058]: Motor head_pan_joint ID=8 has Hardware_Error_Status byte -591776880
[ERROR] [1574104984.338978054]: Motor head_pan_joint ID=8 has Electrical Shock error
[ERROR] [1574104984.339027334]: Motor head_tilt_joint ID=9 has Hardware_Error_Status byte 32765
[ERROR] [1574104984.339070407]: Motor head_tilt_joint ID=9 has Overload error
[ERROR] [1574104984.339111020]: Motor head_tilt_joint ID=9 has Electrical Shock error
[ERROR] [1574104984.339160414]: Motor head_tilt_joint ID=9 has Motor Encoder error
[ERROR] [1574104984.339201517]: Motor head_tilt_joint ID=9 has Overheating error
[ERROR] [1574104984.339253017]: Motor head_tilt_joint ID=9 has Input Volatage error
[ WARN] [1574104984.339470336]: Costmap2DROS transform timeout. Current time: 1574104984.3394, global_pose stamp: 1574104983.6282, tolerance: 0.5000
[ WARN] [1574104984.339515407]: Could not get robot pose, cancelling reconfiguration
[ WARN] [1574104984.366092564]: Map update loop missed its desired rate of 5.0000Hz... the loop actually took 0.6711 seconds
================================================================================REQUIRED process [locobot_controller-11] has died!
process has finished cleanly
log file: /home/locobot/.ros/log/e195d8b4-0a37-11ea-a4b6-48f17fdef7b5/locobot_controller-11*.log
Initiating shutdown!
================================================================================
[move_group-19] killing on exit
[odom_map_broadcaster-15] killing on exit
[map_server-14] killing on exit
[move_base-17] killing on exit
[robot_state_publisher-18] killing on exit
[navigation_velocity_smoother-16] killing on exit
[pyrobot_kinematics-13] killing on exit
[locobot_controller-11] killing on exit
[cmd_vel_mux-10] killing on exit
[calibration_tf_broadcaster-12] killing on exit
[bumper2pointcloud-9] killing on exit
[mobile_base-8] killing on exit
[mobile_base_nodelet_manager-7] killing on exit
[diagnostic_aggregator-6] killing on exit
[camera/points_xyzrgb_hw_registered-5] killing on exit
[camera/color_rectify_color-4] killing on exit
[camera/realsense2_camera-3] killing on exit
[camera/realsense2_camera_manager-2] killing on exit
[ INFO] [1574104985.595536924]: Interrupt Signal is received! Terminating RealSense Node...
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```
Ideally, I would like the robot to run uninterrupted for hours as it collects data. Is there any way to prevent this from happening?

kalyanvasudev, 2019-11-18:
Hi, Looks like your camera motors are shutting down for some reason? 

Also, please make sure their wiring is sturdy and you are not commanding motor angles that lead to collision.

kalyanvasudev, 2019-11-26:
Hi , Were you able to resolve the issue?

luxiangnk, 2019-11-27:
[
[image]
](url)

jgonik, 2019-11-27:
Yes, after double-checking the wiring the issue seems to have gone away. Thanks!

closed 2019-11-27 by jgonik