facebookresearch/pyrobot #56: Improper placement of camera_link frame

swinterbotix, 2020-01-21:
Currently, the camera_link frame is located at the front-center of the D435 as shown below....

[image]

However, according to [this](https://github.com/IntelRealSense/realsense-ros#about-frame-id), the camera_link frame should be located in the depth frame as shown below...

[image]

.. so that the transforms created by the RealSense node to the RGB and other IR sensor line up correctly as shown below...

[image]

kalyanvasudev, 2020-03-05:
@Dhiraj100892 could you please take a loot at this?

kalyanvasudev, 2020-04-10:
@Dhiraj100892  any updates on this?

growupboron, 2020-05-15:
I'll like to work on this, is this still available?

kalyanvasudev, 2020-05-17:
@growupboron , yes, this issue is till up for grabs. If you decide to work on this, please make a pull request onto the Develop branch. Thank you :)

subham9777, 2020-08-10:
Hey @kalyanvasudev, is this issue still opened or is someone working on it?

vkrm1612, 2021-02-13:
Hey i am new to opensource can i fix this issue.....!

Jekyll1021, 2021-02-13:
@vkrm1612 If you are interested, feel free to zip all your changes in a pull request, and assign me as the reviewer!