facebookresearch/pyrobot #58: slam problem in main.launch

yimengli46, 2020-02-09:
Hi, I recently bought a LoCoBot. When I was trying the examples, I run into a problem.
`$ roslaunch locobot_control main.launch use_base:=true use_vslam:=true use_camera:=true`
The error message is,
`[orb_slam2_rgbd-6] process has died [pid 2152, exit code -6, cmd /home/locobot/low_cost_ws/devel/lib/orb_slam2_ros/orb_slam2_rgbd __name:=orb_slam2_rgbd __log:=/home/locobot/.ros/log/39afd322-4b12-11ea-86ed-48f17fdf7f9b/orb_slam2_rgbd-6.log].
log file: /home/locobot/.ros/log/39afd322-4b12-11ea-86ed-48f17fdf7f9b/orb_slam2_rgbd-6*.log`

As a result, I cannot run the navigation examples with slam involved.

Do I need to reinstall PyRobot? Because PyRobot is pre-installed on LoCoBot so I didn't do anything with it.

kalyanvasudev, 2020-02-13:
Yes, please purge and try re-installing again with the official instructions.
Let me know if doesn't resolve the issue.

swinterbotix, 2020-02-26:
@yimengli46,

Were you SSHing into the robot when running that command? If you were, did you make sure to add the '-X' flag? I've seen a similar error to what you are describing, and it resolves itself when you do this. The '-X' flag allows graphical applications on the robot computer (such as orb slam) to work. Otherwise, those applications will just crash and burn.

kalyanvasudev, 2020-03-05:
@yimengli46  I take it that issue is resolved? Please re-open the issue if you have further questions.

closed 2020-03-05 by kalyanvasudev

fujin678, 2020-06-24:
Hi,
i have the same issue, even with "-X" in ssh.     So wonder how the issue was resolved?

thanks,
Jeffrey

fujin678, 2020-06-25:
my issue is fixed.   sorry for any confusion

srama2512, 2022-07-05:
@fujin678 - how did you fix the issue? I'm running into the same problem.