facebookresearch/pyrobot #61: [Feature Contribution] Tests for a custom robot

mbusy, 2020-02-26:
Hi, I started to integrate a custom robot (Pepper robot) in PyRobot a while ago, and I ultimately wish to submit this work through a PR.

Most of the integration is already done, but I do have a question regarding the tests. Would it be better to modify the existing test files (filtering which tests should be launched according to the __botname__ parameter), or to simply to add new test files for the Pepper robot ?

kalyanvasudev, 2020-03-05:
Hi, Please create a separate set of tests for the Pepper robot by making a folder inside the "tests" folder.

mbusy, 2020-03-17:
Hi @kalyanvasudev, thanks for the precision!

kalyanvasudev, 2020-04-10:
I hope this answers your question. Closing this issue for now. Please feel free to re-open it if you have any further questions.

closed 2020-04-10 by kalyanvasudev

mbusy, 2020-04-10:
It does thanks, I will submit a PR as soon as I get an answer from my company regarding the CLA.