facebookresearch/pyrobot #62: Python 3 Installation Issues

swinterbotix, 2020-03-06:
During the installation of Pyrobot (with Python 3), the locobot_install_all.sh script executes this command `pip3 install --ignore-installed -r requirements_python3.txt`.

While it's installing packages, an error message displays saying that the required Numpy version for 'bezier' is at least 1.18.1. However, the [requirements_python3.txt](https://github.com/facebookresearch/pyrobot/blob/master/robots/LoCoBot/requirements_python3.txt) file specifies that it must be equal to 1.17.0. I manually changed this to be 1.18.1 and the problem went away. I believe this occurred since the newest bezier package was released February 20th of this year.

On a different note, when trying to run the camera calibration script (in one terminal I type `roslaunch locobot_calibration calibrate.launch base:=kobuki` and in the second I type `python scripts/collect_calibration_data.py \
    --data_dir tmp/calibration-data/v1/ \
    --botname locobot`
after activating the virtual environment (`load_pyrobot_env`)). However, I get an error message of...

`Traceback (most recent call last):
  File "scripts/collect_calibration_data.py", line 18, in <module>
    import cv2
ImportError: /opt/ros/kinetic/lib/python2.7/dist-packages/cv2.so: undefined symbol: PyCObject_Type`.

This confuses me... if the python 3 environment was activated, why is python looking at python2 packages?!?!?!?

Could you help me fix this issue? I'm in the middle of creating a new SSD image with the most up-to-date PyRobot repo and Python3 (as Python2 is no longer supported) to go with the locobot platforms that we are distributing.

Thanks!

kalyanvasudev, 2020-03-18:
Hi, all of these issues have now been resolved on the Develop branch. Soon I will merge them onto Master after some testing. Thanks for bringing this to my attension.

Regarding the the `cv2` import in python3, with ROS installed, there is no other way to do it than the one you mentioned. I made it look clean using this function - https://github.com/facebookresearch/pyrobot/blob/Develop/src/pyrobot/utils/util.py#L17

So, if you want to perform  `import cv2` in python3 with ROS installed, please use the follow snippet,
```
from pyrobot.utils.util import try_cv2_import
cv2 = try_cv2_import()
```

swinterbotix, 2020-03-18:
Sounds great! Looking forward to trying this out soon once you merge to master.

swinterbotix, 2020-03-30:
Any idea on when the issue that's causing the 'checks to fail' will be resolved?

kalyanvasudev, 2020-04-10:
Hi The checks that are failing are benign controller errors in Gazebo. 
Also, now the master is updated to reflect changes from Develop. So, all the python3 install issues and CV2 errors are resolved.

Please let me know if you have any further questions. Closing this issue for now.

closed 2020-04-10 by kalyanvasudev

swinterbotix, 2020-04-12:
Amazing! Thanks so much. I'll try doing a clean install with the Python3 Version soon and let you know if any other issues come up.