facebookresearch/pyrobot #68: no transform from [map] to [base_link] pyrobot

behzadhaghgoo, 2020-04-14:
I'm trying to run locobot in simulation but have been stuck on this bug for a while. Would be great if you can help. I installed pyrobot on python3 and when I run 
`roslaunch locobot_control main.launch use_base:=true use_sim:=true`
I get 
`[ WARN] [1586887842.616127497, 278.185000000]: Timed out waiting for transform from base_link to map to become available before running costmap, tf error: Could not find a connection between 'map' and 'base_link' because they are not part of the same tree.Tf has two or more unconnected trees.. canTransform returned after 0.109 timeout was 0.1.`
And here is the transformations tree:
[image]

kalyanvasudev, 2020-04-14:
Hi, Could you please provide more information on how you installed it? 

It seems to working on the CI for the master branch. Here are the details,
https://circleci.com/gh/facebookresearch/pyrobot/34?utm_campaign=vcs-integration-link&utm_medium=referral&utm_source=github-build-link

behzadhaghgoo, 2020-04-14:
I used `./locobot_install_all.sh -t full -p 3`. However, when I ran it again it says 
`[ WARN] [1586899635.553950168]: No RealSense devices were found!
... logging to /home/behzad/.ros/log/b545d326-7e96-11ea-86cb-000c2971123e/roslaunch-ubuntu-17213.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:38021/

SUMMARY
========

PARAMETERS
 * /orb_slam2_gen_cfg/bRGB: True
 * /orb_slam2_gen_cfg/camBaseline: 50
 * /orb_slam2_gen_cfg/camInfoTopic: /camera/color/cam...
 * /orb_slam2_gen_cfg/filename: /home/behzad/low_...
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    orb_slam2_gen_cfg (orb_slam2_ros/orb_slam2_gen_cfg)

ROS_MASTER_URI=http://localhost:11311

process[orb_slam2_gen_cfg-1]: started with pid [17230]
[camera/realsense2_camera-3] killing on exit
[camera/realsense2_camera_manager-2] killing on exit
[orb_slam2_gen_cfg-1] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
******************************************
Camera configuration file not generated!!!
Please manually generate it
******************************************
shutting down processing monitor...
... shutting down processing monitor complete
done`
is this the reason? if yes can you please tell me what I should do?

kalyanvasudev, 2020-04-16:
Sorry for the delay, what s the `roslauch` command you are running?

kalyanvasudev, 2020-04-22:
Hi @behzadhaghgoo, were you able to resolve the issue? Let me know if you are still struck.

behzadhaghgoo, 2020-04-26:
I am very sorry for the delay, I thought I responded, 

Yeah, I solved the problem by reinstalling everything from the beginning. Then I just needed to update the gazebo to work. 

Thanks!

kalyanvasudev, 2020-04-27:
Glad to hear that!

closed 2020-04-27 by kalyanvasudev