facebookresearch/pyrobot #69: Odom map broadcaster

fxia22, 2020-04-16:
This line seems to conflict with any other localization packages that tries to generate map -> odom transform right? I am wondering is this intended and what is the best way to avoid the conflict? (I can always comment it out but just wonder) thanks!

https://github.com/facebookresearch/pyrobot/blob/d3ca3c681ce60243267f7bf5650bce8ec0fe177a/robots/LoCoBot/locobot_navigation/base_navigation/launch/move_base.launch#L10

kalyanvasudev, 2020-04-16:
The easiest fix would be change the frame name `map` to something else either in all movebase launch files and config files or in your localization package.

kalyanvasudev, 2020-04-23:
I hope my previous reply answers the question? Please re-open the issue if the issue persists.

closed 2020-04-23 by kalyanvasudev