facebookresearch/pyrobot #70: Realsense Camera Version

swinterbotix, 2020-04-17:
Hi,

When the most current locobot_install_all.sh script installs the librealsense2-dkms version, I get an error message essentially saying that librealsense2-dkms 1.3.4-0ubuntu1:librealsense2-dkms kernel module failed to build'. Currently, my kernel version is 4.15.0-96-generic (latest to date), and I'm working with Ubuntu 16.04.6. As a result, the realsense node is not working.

Is it possible that since you are installing librealsense 2.18.1 (which is outdated by more than a year), it is not able to build correctly on my kernel? I noticed that if I install the latest librealsense2 packages and the realsense_ros node as outlined at https://github.com/IntelRealSense/realsense-ros, that everything works as it should. I also noticed that version 2.18.1 is recommended to work with an outdated realsense firmware version (5.10.03). As Intel makes these cameras, they are putting the latest firmware on them by default (currently 5.11).

Would it be possible to update the installation script such that it always builds the latest stable version of librealsense and the realsense ros node?

Thanks

kalyanvasudev, 2020-04-20:
Hi, thank you for the detailed analysis.

I am looking into the install script issues now. Once after testing, I will update it with the suggested changes. Please consider making [pull requests](https://help.github.com/en/github/collaborating-with-issues-and-pull-requests/creating-a-pull-request) directly in the future instead of attaching the files to the issue. This will give you the direct control over how you would like to add your code to the PyRobot repo :) 

I will also fix the `pickle` error in the `solve_for_calibration_params.py` shortly

kalyanvasudev, 2020-04-20:
It would also be nice to incorporate the trossen-locobot's URDF into this pyrobot repository. Just like `locobot`, we could call it `trossen_locobot` in the botnames. We would also have to incorporate this as an option in [this launch file](https://github.com/facebookresearch/pyrobot/blob/master/robots/LoCoBot/locobot_control/launch/main.launch).

swinterbotix, 2020-04-20:
> I am looking into the install script issues now. Once after testing, I will update it with the suggested changes. Please consider making pull requests directly in the future instead of attaching the [...]

When I started writing the issue, I wasn't planning on attaching anything - it kind of just happened. But I'll do that next time.

> It would also be nice to incorporate the trossen-locobot's URDF into this pyrobot repository. Just like locobot, we could call it trossen_locobot in the botnames. We would also have to incorporate this as an option in this launch file.

Sounds good. I'll create a PR for this soon.

swinterbotix, 2020-04-21:
About that PR...

I've been doing some digging into the locobot packages and there's one file at https://github.com/facebookresearch/pyrobot/blob/master/examples/sim2real/envs/locobotEnv/gym_env.py#L67 that I'm not sure how to fix as it does a pretty good job of hardcoding the original locobot_description.urdf file into it. Any ideas?

Also, I thought of an alternative approach that would not involve creating a bunch of launch file arguments in multiple launch files. What if we named the original locobot_description file a different name - say locobot_cmu_description, and then placed another one next to it called locobot_interbotix_description. Then, as part of the installation procedure, you would specify which one you wanted, and the installation script would create a hard link called locobot_description that would point to one of those two files? That way, no launch files would have to be changed, and things are a bit easier to manage.

kalyanvasudev, 2020-04-21:
> Also, I thought of an alternative approach that would not involve creating a bunch of launch file arguments in multiple launch files. What if we named the original locobot_description file a [...]

I really like this approach :) This way we keep the launch files consistent and simple. This also answers your earlier question elegantly.

swinterbotix, 2020-04-21:
Cool. We'll still need to have it in the main.launch file just the one time though since I specify a different arm sleep pose in the locobot_controller.cpp file. Anyway, I'll go ahead and create a PR for this soon.

swinterbotix, 2020-04-23:
I figured out a way to specify the sleep pose in the locobot_controller.cpp file without having to add a launch file argument... In the installation script, I just add a command to modify that file with the appropriate joint values based on the chosen robot.

In any event, I've submitted a PR that addresses all the changes we discussed and it passes!

kalyanvasudev, 2020-04-25:
Awesome, thank you very much for working on this. I will review and accept the PR.