facebookresearch/pyrobot #71: Python3 Runtime Issues/Warnings

swinterbotix, 2020-04-19:
Platform is Ubuntu 16.04.6 ROS Kinetic using latest pyrobot code.

In one terminal, I launch...

`roslaunch locobot_control main.launch use_base:=true use_arm:=true use_camera:=true`

In a second terminal, I navigate to the manipulation directory for the locobot, and type (after sourcing the workspace)

`python ee_pose_control.py` or any other script.

Every time, I get a couple of warnings saying...

[image]

MoveIt also fails more often than not to plan. I believe this can be remedied by changing the default planner to RRTConnectkConfigDefault [here](https://github.com/facebookresearch/pyrobot/blob/master/robots/LoCoBot/locobot_moveit_config/config/ompl_planning.yaml#L124).

kalyanvasudev, 2020-04-20:
I've resolved the warnings on the Develop branch. Will push the changes to master after confirming.

On the other hand, moveit issue a rather interesting one. Let me investigate it more.

kalyanvasudev, 2020-04-23:
Looks like all the warnings are resolved now. I've also updates the moveit default planner to RRTConnectkConfigDefault. 

Closing this issue. Please re-open it if the issue persists.

closed 2020-04-23 by kalyanvasudev