facebookresearch/pyrobot #73: The depth image is NaN

y406539259, 2020-04-21:
## Steps to reproduce
from pyrobot import Robot
import cv2
bot = Robot('locobot')

bot.camera.reset()
camera_pose = [0, 0.7]
bot.camera.set_pan_tilt(camera_pose[0], camera_pose[1], wait=True)

rgb, depth = bot.camera.get_rgb_depth()
cv2.imshow('Color', rgb[:, :, ::-1])
cv2.imshow('Depth', depth)
cv2.waitKey(3000)

## Observed Results

rgb, depth = bot.camera.get_rgb_depth()
rgb image is correctly acquired but the depth image is NaN

y406539259, 2020-04-21:
The Ubuntu Version is 16.04,  in simulator with python=2.7.
There were no error when i ran 
roslaunch locobot_control main.launch use_base:=true use_camera:=true \
>     use_arm:=false use_sim:=true base:=kobuki use_rviz:=false

The depth image is being published like this:
0,0,192,127,0,0,192,127,0,0,192,127
0,0,192,127,0,0,192,127,0,0,192,127
0,0,192,127,0,0,192,127,0,0,192,127

but  
self.depth_img = self.cv_bridge.imgmsg_to_cv2(depth, "passthrough") gets NaN

kalyanvasudev, 2020-04-22:
Please delete the `low_cost_ws` and `pyenv_pyrobot_python2` directories. Then curl get the latest install script and install it using the instructions mentioned [here](https://github.com/facebookresearch/pyrobot/blob/master/README.md) and test it out again.

Also, this error doesn't seem to occurring in the CircleCi tests running [here](https://app.circleci.com/pipelines/github/facebookresearch/pyrobot/92/workflows/6a6a7de2-2c92-4fbc-84f8-ca5c043161a6/jobs/122). Please go though the full test summary for more details.

y406539259, 2020-04-24:
~/low_cost_ws/src/pyrobot/tests$ pytest test_camera.py -v

============================= test session starts ==============================
platform linux2 -- Python 2.7.12, pytest-4.6.9, py-1.8.1, pluggy-0.13.1 -- /usr/bin/python
cachedir: .pytest_cache
metadata: {'Python': '2.7.12', 'Platform': 'Linux-4.15.0-96-generic-x86_64-with-Ubuntu-16.04-xenial', 'Packages': {'py': '1.8.1', 'pytest': '4.6.9', 'pluggy': '0.13.1'}, 'Plugins': {u'html': u'1.20.0', u'metadata': u'1.8.0', u'cov': u'2.6.1'}}
rootdir: /home/yinpengshuai/low_cost_ws/src/pyrobot
plugins: html-1.20.0, cov-2.6.1, metadata-1.8.0
collected 6 items                                                              

test_camera.py::test_position_control[target_position0] PASSED           [ 16%]
test_camera.py::test_position_control[target_position1] PASSED           [ 33%]
test_camera.py::test_position_control[target_position2] PASSED           [ 50%]
test_camera.py::test_position_control[target_position3] PASSED           [ 66%]
test_camera.py::test_position_control[target_position4] PASSED           [ 83%]
test_camera.py::test_get_images PASSED                                   [100%]

=========================== 6 passed in 4.35 seconds ===========================

But in Gazebot simulator,  Depth image is NaN, RGB image is normal.

y406539259, 2020-04-24:
The depth image is not none, the depth image size is 480x640 and each pixel value of the depth image is nan.

"assert depth_img is not None"  success but the pixel value of the depth image is nan.

kalyanvasudev, 2020-04-25:
I've adjust the near and far plane clipping values for the depth camera in [this commit latest commit](https://github.com/facebookresearch/pyrobot/commit/c44d901b0c72afd00cda3cab0b9b9d236a02ebd5). So, not all the values will be NaN.

Basically, the pixels with the value NaN mean that the depth value is outside near and far clipping planes. 

I hope this helps. Please feel free post any further questions you have the same aspect.

kalyanvasudev, 2020-05-08:
I hope this has resolved your issues. Please feel free to re-open this issue if you have any further questions. Thanks!

closed 2020-05-08 by kalyanvasudev