facebookresearch/pyrobot #74: Habitat integration broken

soumith, 2020-04-23:
After commit https://github.com/facebookresearch/pyrobot/commit/73a741c9ca040838fc39b42b654875051f3d9f5a which refactored the code to add Kinect2 integration, Habitat integration is broken.

This is because, the core camera class [now expects](https://github.com/facebookresearch/pyrobot/blob/7c3f0767f6678a4eb1b296adef5261670d6db096/src/pyrobot/core.py#L336-L344) all yacs configs to have attributes like `CAMERA.ROSTOPIC_CAMERA_INFO_STREAM`, but Habitat's yacs config doesn't have it.

The branch right before this commit has working Habitat integration

kalyanvasudev, 2020-04-24:
Hi, I've fixed it in the latest commits on the master.

kalyanvasudev, 2020-04-24:
I've also added some tests for it in the test folder and also updated the runtime instructions for the habitat example. 
However, I'd some trouble enabling tests for it on the circleci as habitat in headless mode relies on particular environment configuration which has Nvidia Cuda and OpenGL dependencies. This seems doable with the `nvidia/cudagl` docker as the base in Ubuntu 18.04. However, I don't have a clear solution other than mahcine executor for Ubuntu 16.04. I will add this to the circleCI too when I have clean solution

soumith, 2020-04-24:
feel free to use the config I wrote in https://github.com/fairinternal/minecraft/pull/690/files for Habitat CircleCI environment

soumith, 2020-04-24:
thanks for the quick fix

closed 2020-04-24 by soumith