facebookresearch/pyrobot #81: Remote Networking Setup

swinterbotix, 2020-05-06:
Hi,

I noticed in your documentation for setting up a Locobot to work with a remote machine [here](https://pyrobot.org/docs/network), you specify that...

> So you will run all the robot driver or PyRobot code on your local machine (hence, it can be much faster and it also means that you should have all the software setup in your local machine), and then the commands are sent to the NUC machine to control the robot.

When you say 'all the software setup in your local machine', does that mean I should run the 'locobot_install_all.sh' script on my local machine? If so, what should I do about the fact that it says the RealSense camera must be plugged in? Should I just plug the camera in to my local computer for the installation, and then plug it back into the locobot computer afterwards?

OR...

Would it be enough to just run the 'install_pyrobot.sh' script on my local machine?

OR...

Should I run the locobot_install_all.sh script on my local machine, but do sim_only? If so, will I still be able to access the real camera from my local computer after networking?

Thanks

kalyanvasudev, 2020-05-08:
Local computer = LoCoBot computer
Remote computer = The ubuntu machine you are using the pyrobot python API on.

Hi @swinterbotix, the ideal setup to work in this regard would be the locobot-full install on the robot local computer and then use pyrobot standalone or locobot-sim_only install method on the remote machine. Post that, use the instructions mentioned [here](https://pyrobot.org/docs/network) to link both the computers. 

Then, the workflow would be to run the roslaunch commands on the robot local machine and the pyrobot python scripts on your remote machine.

Basically, the full install should occur on the computer in which you plug all the robot hardware (base, arm and camera).

swinterbotix, 2020-05-08:
Awesome - thanks

closed 2020-05-08 by swinterbotix