facebookresearch/pyrobot #83: PyRobot Installation Doesn't Work for Ubuntu 18.04 and Python 3

msr-peng, 2020-05-18:
Hey Kalyan, I'm not sure whether PyRobot can also run on Ubuntu 18.04 by Python 3. But I did run issues with installation when execute [locobot_install_all.sh](https://github.com/facebookresearch/pyrobot/blob/master/robots/LoCoBot/install/locobot_install_all.sh) on a fresh Ubuntu 18.04.

To make the installation scripts run without errors, I did 4 minor modifications:

  1. I added `sudo apt-get install python-rosdep` before [Line 134](https://github.com/facebookresearch/pyrobot/blob/master/robots/LoCoBot/install/locobot_install_all.sh#L134), otherwise, there would be `sudo: rosdep: command not found` error.
  2. I deleted `python-software-properties` in [`install_pyrobot.sh` Line 67](https://github.com/facebookresearch/pyrobot/blob/master/install_pyrobot.sh#L67), since `python-software-properties` package is only available for Ubuntu 12-, and `software-properties-common` is its alternative.
  3. I commented out [`install_pyrobot.sh` Line 68](https://github.com/facebookresearch/pyrobot/blob/master/install_pyrobot.sh#L68), otherwise, there would be errors like this:

[image]

  4. For [`install_pyrobot.sh` Line 107](https://github.com/facebookresearch/pyrobot/blob/master/install_pyrobot.sh#L107), instead of using `catkin_make`, I used `catkin_make_isolated`, otherwise would occur errors like this:
[image]

## Observed Results
After installation, when import pyrobot module in PyRobot virtual env this ImportError would occur:
[image]
 
It has something do do with [`client.py` Line 68](https://github.com/kalyanvasudev/ros_comm/blob/melodic-devel/clients/rospy/src/rospy/client.py#L68) and [roscpp.srv](https://github.com/kalyanvasudev/ros_comm/tree/melodic-devel/clients/roscpp/srv) module. I noticed there isn't `GetLoggersResponse`, `SetLoggerLevelResponse` module. So is the installation scripts for Ubuntu 18.04 and Python 3 still unfinished?

kalyanvasudev, 2020-05-19:
Hi @msr-peng, Thanks for bringing this to my attention. Yeah,  I've observed these issues a week back and had resolved them on the Develop branch. 

I will also soon merge the changes from Develop branch onto master in a day.

msr-peng, 2020-05-19:
Oh that's cool. I'll close this issue then.

closed 2020-05-19 by msr-peng