facebookresearch/pyrobot #85: No module named forms.form

wangjiwang, 2020-05-20:
when I run "python locobot.py ", I get the following error.   

(pyenv_locobot_grasping) wjw@wjw-ThinkPad-T480:~/low_cost_ws/src/pyrobot/examples/grasping$ python locobot.py 
Traceback (most recent call last):
  File "locobot.py", line 23, in <module>
    from pyrobot import Robot
  File "/home/wjw/.local/lib/python2.7/site-packages/pyrobot/__init__.py", line 3, in <module>
    from .browser import RoboBrowser
  File "/home/wjw/.local/lib/python2.7/site-packages/pyrobot/browser.py", line 12, in <module>
    from .forms.form import Form
ImportError: No module named forms.form

kalyanvasudev, 2020-05-21:
Hi, Could you please provide more information on how you have installed the Pyrobot and how your are running the example script?

wangjiwang, 2020-05-21:
> Hi, Could you please provide more information on how you have installed the Pyrobot and how you are running the example script?

@kalyanvasudev I ran the script as the link https://www.pyrobot.org/docs/grasping.
and I use the command "pip install pyrobot" to  install Pyrobot.  and I also ran the command "chmod +x locobot_install_all.sh 
./locobot_install_all.sh -t sim_only -p 2"

wangjiwang, 2020-05-21:
the following is what I just did, and the problem is same.  [ I don not have any robot now !!!]
[ open a terminal ]
source ~/pyenv_locobot_grasping/bin/activate
roslaunch locobot_control main.launch use_arm:=true use_camera:=true use_sim:=true

[ open another terminal ]
source ~/pyenv_locobot_grasping/bin/activate
cd ~/low_cost_ws/src/pyrobot/examples/grasping
python locobot.py --n_grasps=5 --n_samples=100 --patch_size=100

then I got the error:

Traceback (most recent call last):
  File "locobot.py", line 23, in <module>
    from pyrobot import Robot
  File "/home/wjw/.local/lib/python2.7/site-packages/pyrobot/__init__.py", line 3, in <module>
    from .browser import RoboBrowser
  File "/home/wjw/.local/lib/python2.7/site-packages/pyrobot/browser.py", line 12, in <module>
    from .forms.form import Form
ImportError: No module named forms.form

kalyanvasudev, 2020-05-28:
Hi, before we test this out, could you please confirm if the examples mentioned in [this](https://www.pyrobot.org/docs/navigation) tutorial are working?

kalyanvasudev, 2020-07-07:
Closing the issue due to inactivity. Please reopen if the issue persists.

closed 2020-07-07 by kalyanvasudev

Engnation, 2021-01-22:
Hello @kalyanvasudev, I ran the install `chmod +x locobot_install_all.sh` `./locobot_install_all.sh -t sim_only -p 3 -l interbotix` script (quite a while ago), and am having the same issue when trying to run: `python run_cmp.py --goal_x 1.2 --goal_y 0.0 --goal_t 0. --botname locobot` from [this tutorial](https://pyrobot.org/docs/visual_nav_cmp). Sorry, the link above in the May 28th post is broken, but I have been able to run many other examples. I realise that I have run the -p 3 install parameter, so I manually created pyenv_pyrobot (containing python 2.7) and ran `pip install -r requirements.txt` and also manually ran `pip install pyrobot` in that pyenv (I also uninstalled pyrobot and tried running `python -m pip install pyrobot`), but I get `ImportError: No module named pyrobot.utils.util` when I run `pytest test_cmp.py` and `ImportError: No module named forms.form` when I run: `python run_cmp.py --goal_x 1.2 --goal_y 0.0 --goal_t 0. --botname locobot `. Any thoughts as to what could be wrong?

quangtnv, 2021-06-09:
I have the same problem when running the scripts :( Any tips?