facebookresearch/pyrobot #89: Runtime Error

swinterbotix, 2020-06-11:
Hi,

I just installed the latest code on my locobot, and while the installation script worked flawlessly (full install, Python 3, interbotix locobot, ros kinetic on Ubuntu 16.04, Intel NUC7i5BNH), I ran into an issue right afterwards.

Specifically, I opened a terminal and typed `roslaunch locobot_control main.launch use_arm:=true use_base:=true use_camera:=true`, and got the following error...

[image]

I noticed (from your commits) that it looked like you were going to include some sort of gpmp action server but it didn't make the final cut. However, it looks like some part of it made it through to the latest commit. Any way that you could take a look at this?

kalyanvasudev, 2020-06-11:
1. I've resolved the runtime error in the latest commit on the master. 

2. Let me look into the ee_pose_control.py script.

swinterbotix, 2020-06-11:
Sounds good! Let me know know what you find

kalyanvasudev, 2020-06-19:
HI @swinterbotix , I've fixed the moveit issue on the Develop branch. Now, planning is more reliable than earlier. I will merger into the master in about 10 days.

closed 2020-06-19 by kalyanvasudev

swinterbotix, 2020-06-19:
Amazing! Looking forward to that.