facebookresearch/pyrobot #90: error code: CONTROL_FAILED  with example app grasping

fujin678, 2020-06-11:
## Steps to reproduce

  1. _____python 2.7
  2. _____locobot
  3. _____

## Observed Results

python locobot.py --n_grasps=1 --n_samples=20 --patch_size=100
[ERROR] [1591915854.422550]: Timeout waiting for Inverse Kinematics Service!!
[ERROR] [1591915857.475432]: Timeout waiting for Forward Kinematics Service!!
[WARN] [1591915857.741032]: Neither use_base, nor use_sim, is not set to True when the LoCoBot driver is launched. You may not be able to command the base correctly using PyRobot!
Loading grasp model
/home/lab/pyenv_locobot_grasping/local/lib/python2.7/site-packages/torch/serialization.py:325: SourceChangeWarning: source code of class 'grasp_samplers.deeper_models.IfullRobHWNet' has changed. Tried to save a patch, but couldn't create a writable file IfullRobHWNet.patch. Make sure it doesn't exist and your working directory is writable.
  warnings.warn(msg, SourceChangeWarning)
Time taken to load model: 0.0426759719849s
[INFO] [1591915860.030768]: Grasp attempt #1
[INFO] [1591915860.031121]: Moveit Motion Planning...
[INFO] [1591915860.147579]: Moveit Failed with error code: CONTROL_FAILED
[INFO] [1591915860.147964]: Moveit Motion Planning...
[INFO] [1591915860.250241]: Moveit Failed with error code: CONTROL_FAILED
[INFO] [1591915860.250680]: Moveit Motion Planning...
[INFO] [1591915860.351707]: Moveit Failed with error code: CONTROL_FAILED
[INFO] [1591915860.352153]: Moveit Motion Planning...
[INFO] [1591915860.469831]: Moveit Failed with error code: CONTROL_FAILED
[INFO] [1591915860.470428]: Moveit Motion Planning...
[INFO] [1591915860.571781]: Moveit Failed with error code:

## Expected Results

  * What did you expect to happen?

## Relevant Code

```
// TODO(you): code here to reproduce the problem
```

kalyanvasudev, 2020-06-14:
Looks like the IK and FK services are not running? Did you run the appropriate roslaunch command before running the python script?

fujin678, 2020-06-15:
thanks for quick response.  my roslaunch is:

roslaunch locobot_control main.launch use_arm:=true use_base:=false use_camera:=true use_rviz:=false

kalyanvasudev, 2020-06-19:
Seems like a moveit issue then. Could you please confirm if the scripts in the example folder work for you? Also, I've improved the moveit performance on the Develop branch. I will soon merge them into master.

closed 2020-07-07 by kalyanvasudev