facebookresearch/pyrobot #92: vis_3d_map.py  does not show any point cloud

fujin678, 2020-06-25:
## Steps to reproduce

  1. _____roslaunch locobot_control main.launch use_arm:=true use_base:=true use_camera:=true use_rviz:=false use_vslam:=true

  2. _____python vis_3d_map.py
  3. _____

## Observed Results
 vis_3d_map.py only shows coordinate axis,  but no point cloud update, although the update while-loop runs every 4sec

further debuggging shows after running "pts, colors = bot.base.base_state.vslam.get_3d_map()",      "pts" and "colors" are always empty

  * What happened?  This could be a description, log output, etc.

## Expected Results

  * What did you expect to happen?

## Relevant Code

```
// TODO(you): code here to reproduce the problem
```

fujin678, 2020-06-25:
slam_get_pose.py works ok.  Does this mean vslam is working?   But somehow, "bot.base.base_state.vslam.get_3d_map()" returns empty array

kalyanvasudev, 2020-07-07:
At the start of the robot, the point cloud is empty. It continues to build and accumulate as the robot moves.

closed 2020-07-07 by kalyanvasudev