facebookresearch/pyrobot #94: import pyrobot error

mumu1408, 2020-07-01:
## Steps to reproduce

mkdir -p ~/code/habitat_navrobot/navrobot_ws/src
cd ~/code/habitat_navrobot/navrobot_ws/src
git clone --recurse-submodules https://github.com/facebookresearch/pyrobot.git
cd pyrobot/
chmod +x install_pyrobot.sh
./install_pyrobot.sh -p 3

## Observed Results

 ```
(pyenv_pyrobot_python3) ee218@ee218-X555LI:~$ python
Python 3.6.2 (default, Jul 17 2017, 23:14:31) 
[GCC 5.4.0 20160609] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyrobot
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/ee218/pyenv_pyrobot_python3/lib/python3.6/site-packages/pyrobot/__init__.py", line 6, in <module>
    from pyrobot.core import Robot
  File "/home/ee218/pyenv_pyrobot_python3/lib/python3.6/site-packages/pyrobot/core.py", line 18, in <module>
    import tf
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/tf/__init__.py", line 28, in <module>
    from tf2_ros import TransformException as Exception, ConnectivityException, LookupException, ExtrapolationException
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/tf2_ros/__init__.py", line 38, in <module>
    from tf2_py import *
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/tf2_py/__init__.py", line 38, in <module>
    from ._tf2 import *
ImportError: dynamic module does not define module export function (PyInit__tf2)
```
## The output of ./install_pyrobot.sh -p 3
```
Ubuntu 16.04 detected. ROS-Kinetic chosen for installation.
Python 3 chosen for pyRobot installation.
Reading package lists...
Building dependency tree...
Reading state information...
python-virtualenv is already the newest version (15.0.1+ds-3ubuntu1).
0 upgraded, 0 newly installed, 0 to remove and 451 not upgraded.
Reading package lists...
Building dependency tree...
Reading state information...
ros-kinetic-kdl-parser-py is already the newest version (1.12.11-0xenial-20191214-075722+0000).
ros-kinetic-orocos-kdl is already the newest version (1.3.2-1xenial-20191214-034014+0000).
ros-kinetic-python-orocos-kdl is already the newest version (1.3.2-1xenial-20191214-035826+0000).
ros-kinetic-trac-ik is already the newest version (1.5.1-1xenial-20200312-224142+0000).
0 upgraded, 0 newly installed, 0 to remove and 451 not upgraded.
Reading package lists...
Building dependency tree...
Reading state information...
software-properties-common is already the newest version (0.96.20.9).
0 upgraded, 0 newly installed, 0 to remove and 451 not upgraded.
Reading package lists...
Building dependency tree...
Reading state information...
python-software-properties is already the newest version (0.96.20.9).
0 upgraded, 0 newly installed, 0 to remove and 451 not upgraded.
OK
Hit:1 http://mirrors.zju.edu.cn/ubuntu xenial InRelease
Hit:2 http://mirrors.zju.edu.cn/ubuntu xenial-security InRelease
Hit:3 http://mirrors.zju.edu.cn/ubuntu xenial-updates InRelease
Hit:4 http://mirrors.zju.edu.cn/ubuntu xenial-backports InRelease
Hit:5 http://packages.ros.org/ros/ubuntu xenial InRelease
Hit:6 http://packages.microsoft.com/repos/vscode stable InRelease
Hit:7 http://ppa.launchpad.net/fkrull/deadsnakes/ubuntu xenial InRelease
Hit:8 http://ppa.launchpad.net/ubuntu-toolchain-r/test/ubuntu xenial InRelease
Reading package lists...
Reading package lists...
Building dependency tree...
[... 440 lines cut ...]
-- Found the following Boost libraries:
--   signals
--   system
--   thread
--   chrono
--   date_time
--   atomic
-- +++ processing catkin package: 'tf2_bullet'
-- ==> add_subdirectory(geometry2/tf2_bullet)
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'bullet'
--   No package 'bullet' found
-- Configuring incomplete, errors occurred!
See also "/home/ee218/pyrobot_catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/ee218/pyrobot_catkin_ws/build/CMakeFiles/CMakeError.log".
Base path: /home/ee218/pyrobot_catkin_ws
Source space: /home/ee218/pyrobot_catkin_ws/src
Build space: /home/ee218/pyrobot_catkin_ws/build
Devel space: /home/ee218/pyrobot_catkin_ws/devel
Install space: /home/ee218/pyrobot_catkin_ws/install
Creating symlink "/home/ee218/pyrobot_catkin_ws/src/CMakeLists.txt" pointing to "/opt/ros/kinetic/share/catkin/cmake/toplevel.cmake"
####
#### Running command: "cmake /home/ee218/pyrobot_catkin_ws/src -DPYTHON_EXECUTABLE=/home/ee218/pyenv_pyrobot_python3/bin/python -DPYTHON_INCLUDE_DIR=/usr/include/python3.6m -DPYTHON_LIBRARY=/usr/lib/x86_64-linux-gnu/libpython3.6m.so -DCATKIN_DEVEL_PREFIX=/home/ee218/pyrobot_catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/ee218/pyrobot_catkin_ws/install -G Unix Makefiles" in "/home/ee218/pyrobot_catkin_ws/build"
####
```

kalyanvasudev, 2020-07-01:
Things seem to running smoothly here on the CI - https://app.circleci.com/pipelines/github/facebookresearch/pyrobot/216/workflows/23884249-eaaa-4423-9dd8-e0b8e7e9b165/jobs/397/steps

What is the output of the command `echo $PYTHONPATH`?

kalyanvasudev, 2020-07-01:
Also, is the that the full output log of the python3-pyrobot installation? Seems the the `pyrobot_catkin_ws` is not fully compiled?

mumu1408, 2020-07-02:
> Things seem to running smoothly here on the CI - https://app.circleci.com/pipelines/github/facebookresearch/pyrobot/216/workflows/23884249-eaaa-4423-9dd8-e0b8e7e9b165/jobs/397/steps
> 
> What is the output of the command `echo $PYTHONPATH`?

```
(pyenv_pyrobot_python3) ee218@ee218-X555LI:~$ echo $PYTHONPATH
/home/ee218/code/turtlebot/turtlebot/devel/lib/python2.7/dist-packages:/home/ee218/code/turtlebot/kobuki/devel/lib/python2.7/dist-packages:/home/ee218/code/turtlebot/rocon/devel/lib/python2.7/dist-packages:/opt/ros/kinetic/lib/python2.7/dist-packages
```

mumu1408, 2020-07-02:
> Also, is the that the full output log of the python3-pyrobot installation? Seems the the `pyrobot_catkin_ws` is not fully compiled?

Yes, that is the full output log.

mumu1408, 2020-07-02:
> Also, is the that the full output log of the python3-pyrobot installation? Seems the the `pyrobot_catkin_ws` is not fully compiled?

I have re-compiled the 'pyrobot_catkin_ws', but I still have the import error

## the output of 'catkin_make'
```
(pyenv_pyrobot_python3) ee218@ee218-X555LI: ~/pyrobot_catkin_ws$ catkin_make --cmake-args -DPYTHON_EXECUTABLE=$(which python) -DPYTHON_INCLUDE_DIR=/usr/include/python3.6m -DPYTHON_LIBRARY=/usr/lib/x86_64-linux-gnu/libpython3.6m.so
Base path: /home/ee218/pyrobot_catkin_ws
Source space: /home/ee218/pyrobot_catkin_ws/src
Build space: /home/ee218/pyrobot_catkin_ws/build
Devel space: /home/ee218/pyrobot_catkin_ws/devel
Install space: /home/ee218/pyrobot_catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/ee218/pyrobot_catkin_ws/build"
####
####
#### Running command: "make -j4 -l4" in "/home/ee218/pyrobot_catkin_ws/build"
####
[  0%] Built target rosconsole_backend_interface
[  0%] Built target geometry_msgs_generate_messages_nodejs
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target roslz4
[  0%] Built target std_msgs_generate_messages_nodejs
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target geometry_msgs_generate_messages_cpp
[  0%] Built target geometry_msgs_generate_messages_py
[  0%] Built target geometry_msgs_generate_messages_lisp
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target std_msgs_generate_messages_eus
[  0%] Built target geometry_msgs_generate_messages_eus
[  0%] Built target _test_roslib_comm_generate_messages_check_deps_FillEmbedTime
[  0%] Built target _test_roslib_comm_generate_messages_check_deps_TypeNameChangeComplex2
[  0%] Built target _test_roslib_comm_generate_messages_check_deps_FillSimple
[  0%] Built target _test_roslib_comm_generate_messages_check_deps_FieldNameChange1
[  0%] Built target rosgraph_msgs_generate_messages_cpp
[  0%] Built target _test_roslib_comm_generate_messages_check_deps_TypeNameChangeComplex1
[  0%] Built target _test_roslib_comm_generate_messages_check_deps_SameSubMsg1
[  0%] Built target _test_roslib_comm_generate_messages_check_deps_FieldNameChange2
[  0%] Built target _test_roslib_comm_generate_messages_check_deps_HeaderTest
[  0%] Built target _test_roslib_comm_generate_messages_check_deps_TypeNameChangeArray2
[  0%] Built target _test_roslib_comm_generate_messages_check_deps_SameSubMsg2
[  0%] Built target _test_roslib_comm_generate_messages_check_deps_TypeNameChange1
[  0%] Built target _test_roslib_comm_generate_messages_check_deps_SameSubMsg3
[  0%] Built target rosgraph_msgs_generate_messages_nodejs
[  0%] Built target rosgraph_msgs_generate_messages_lisp
[  0%] Built target _test_roslib_comm_generate_messages_check_deps_TypeNameChange2
[  0%] Built target rosgraph_msgs_generate_messages_eus
[  0%] Built target _test_roslib_comm_generate_messages_check_deps_TypeNameChangeArray1
[  0%] Built target _test_roslib_comm_generate_messages_check_deps_ArrayOfMsgs
[  0%] Built target rosgraph_msgs_generate_messages_py
[  0%] Built target _tf2_msgs_generate_messages_check_deps_LookupTransformActionResult
[  0%] Built target _tf2_msgs_generate_messages_check_deps_FrameGraph
[  0%] Built target _tf2_msgs_generate_messages_check_deps_TFMessage
[  0%] Built target _tf2_msgs_generate_messages_check_deps_LookupTransformGoal
[  0%] Built target actionlib_msgs_generate_messages_nodejs
[  0%] Built target _tf2_msgs_generate_messages_check_deps_LookupTransformActionGoal
[  0%] Built target _tf2_msgs_generate_messages_check_deps_LookupTransformAction
[  0%] Built target _tf2_msgs_generate_messages_check_deps_TF2Error
[  0%] Built target _tf2_msgs_generate_messages_check_deps_LookupTransformResult
[  0%] Built target actionlib_msgs_generate_messages_cpp
[  0%] Built target actionlib_msgs_generate_messages_lisp
[  0%] Built target actionlib_msgs_generate_messages_py
[  0%] Built target actionlib_msgs_generate_messages_eus
[  0%] Built target _tf2_msgs_generate_messages_check_deps_LookupTransformActionFeedback
[  1%] Built target xmlrpcpp
[  1%] Built target _tf2_msgs_generate_messages_check_deps_LookupTransformFeedback
[  1%] Built target _roscpp_generate_messages_check_deps_Logger
[  1%] Built target _roscpp_generate_messages_check_deps_SetLoggerLevel
[  1%] Built target _catkin_empty_exported_target
[  1%] Built target _roscpp_generate_messages_check_deps_GetLoggers
[... 213 lines cut ...]
[100%] Built target tf_conversions
```

## the error
```
(pyenv_pyrobot_python3) ee218@ee218-X555LI: ~/pyrobot_catkin_ws$ python
Python 3.6.2 (default, Jul 17 2017, 23:14:31) 
[GCC 5.4.0 20160609] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import tf
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/tf/__init__.py", line 28, in <module>
    from tf2_ros import TransformException as Exception, ConnectivityException, LookupException, ExtrapolationException
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/tf2_ros/__init__.py", line 38, in <module>
    from tf2_py import *
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/tf2_py/__init__.py", line 38, in <module>
    from ._tf2 import *
ImportError: dynamic module does not define module export function (PyInit__tf2)
```
## the output of command `echo $PYTHONPATH`
```
ee218@ee218-X555LI: ~$ echo $PYTHONPATH
/home/ee218/code/habitat_navrobot/navrobot_ws/devel/lib/python2.7/dist-packages:/opt/ros/kinetic/lib/python2.7/dist-packages
```

mumu1408, 2020-07-02:
Just source ~/pyrobot_catkin_ws/devel/setup.bash, and the error gone.
```
(pyenv_pyrobot_python3) ee218@ee218-X555LI: ~/pyrobot_catkin_ws$ source ~/pyrobot_catkin_ws/devel/setup.bash 
(pyenv_pyrobot_python3) ee218@ee218-X555LI: ~/pyrobot_catkin_ws$ python
Python 3.6.2 (default, Jul 17 2017, 23:14:31) 
[GCC 5.4.0 20160609] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import tf
>>> tf.__path__
['/home/ee218/pyrobot_catkin_ws/devel/lib/python3/dist-packages/tf', '/home/ee218/pyrobot_catkin_ws/src/geometry/tf/src/tf', '/opt/ros/kinetic/lib/python2.7/dist-packages/tf']
>>> import sys
>>> sys.path
['/home/ee218/pyrobot_catkin_ws/src/ros_comm/tools/rosgraph/src', '/home/ee218/pyrobot_catkin_ws/src/ros_comm/clients/rospy/src', '/home/ee218/pyrobot_catkin_ws/src/geometry2/tf2_py/src', '/home/ee218/pyrobot_catkin_ws/src/geometry2/tf2_ros/src', '/home/ee218/pyrobot_catkin_ws/src/geometry/tf/src', '', '/home/ee218/pyrobot_catkin_ws/devel/lib/python3/dist-packages', '/home/ee218/code/habitat_navrobot/navrobot_ws/devel/lib/python2.7/dist-packages', '/opt/ros/kinetic/lib/python2.7/dist-packages', '/home/ee218/pyenv_pyrobot_python3/lib/python36.zip', '/home/ee218/pyenv_pyrobot_python3/lib/python3.6', '/home/ee218/pyenv_pyrobot_python3/lib/python3.6/lib-dynload', '/usr/lib/python3.6', '/home/ee218/pyenv_pyrobot_python3/lib/python3.6/site-packages']
>>> import pyrobot
>>> 
```
So, it was my carelessness not to `load_pyrobot_env`

closed 2020-07-02 by mumu1408