facebookresearch/pyrobot #96: tf error: Could not find a connection between 'map' and 'base_footprint' because they are not part of the same tree.Tf has two or more unconnected trees.. canTransform returned after 0.100597 timeout was 0.1. in real locobot enviornment

priyashkla, 2020-07-02:
## Steps to reproduce

1.  In one terminal, type ‘roslaunch locobot_control main.launch use_base:=true use_arm:=true use_camera:=true use_rviz:=false. 
2. In a second terminal, type ‘source ~/pyenv_pyrobot/bin/activate’ and hit Enter. Then navigate to the low_cost_ws/src/pyrobot/examples/locobot directory, and run 
python navigation/camera_image.py

## Observed Results
**first terminal**
 [ WARN] [1593657605.971560081]: The complete state of the robot is not yet known.  Missing wheel_left_joint, wheel_right_joint
[ WARN] [1593657607.123305077]: The complete state of the robot is not yet known.  Missing wheel_left_joint, wheel_right_joint
[ WARN] [1593657608.156443146]: The complete state of the robot is not yet known.  Missing wheel_left_joint, wheel_right_joint
[ WARN] [1593657609.156806335]: The complete state of the robot is not yet known.  Missing wheel_left_joint, wheel_right_joint
[ WARN] [1593657610.227221656]: The complete state of the robot is not yet known.  Missing wheel_left_joint, wheel_right_joint
[ WARN] [1593657610.467207508]: Timed out waiting for transform from base_footprint to map to become available before running costmap, tf error: Could not find a connection between 'map' and 'base_footprint' because they are not part of the same tree.Tf has two or more unconnected trees.. canTransform returned after 0.100731 timeout was 0.1.
**Second terminal**
(pyenv_pyrobot) locobot@locobot:~/low_cost_ws/src/pyrobot/examples/locobot$ python navigation/camera_image.py
[ INFO] [1593657538.427438486]: Loading robot model 'locobot'...
[ INFO] [1593657538.427505010]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1593657538.863965620]:(pyenv_pyrobot) locobot@locobot:~/low_cost_ws/src/pyrobot/examples/locobot/navigation$ cd .. IK Using joint shoulder_link -1.57 1.57
[ INFO] [1593657538.863993998]: IK Using joint elbow_link -1.57 1.57
[ INFO] [1593657538.864010939]: IK Using joint forearm_link -1.57 1.57
[ INFO] [1593657538.864022995]: IK Using joint wrist_link -1.57 1.57
[ INFO] [1593657538.864032792]: IK Using joint gripper_link -1.57 1.57
[ INFO] [1593657538.864046149]: Looking in common namespaces for param name: arm/position_only_ik
[ INFO] [1593657538.865586838]: Looking in common namespaces for param name: arm/solve_type
[ INFO] [1593657538.867052729]: Using solve type Speed
[ INFO] [1593657539.869189503]: Ready to take commands for planning group arm.
Unknown attribute: name
Traceback (most recent call last):
  File "navigation/camera_image.py", line 16, in <module>
    bot = Robot('locobot')
  File "/home/locobot/pyenv_pyrobot/local/lib/python2.7/site-packages/pyrobot/core.py", line 108, in __init__
    self.base = base_class(self.configs, **base_config)
  File "/home/locobot/pyenv_pyrobot/local/lib/python2.7/site-packages/pyrobot/locobot/base.py", line 239, in __init__
    self.planner = MoveBasePlanner(self.configs)
  File "/home/locobot/pyenv_pyrobot/local/lib/python2.7/site-packages/pyrobot/locobot/base_control_utils.py", line 256, in __init__
    rospy.wait_for_service(self.configs.BASE.PLAN_TOPIC, timeout=3)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 143, in wait_for_service
    raise ROSException("timeout exceeded while waiting for service %s"%resolved_name)
rospy.exceptions.ROSException: timeout exceeded while waiting

## Expected Results

  It should give image.

## Relevant Code

```
// TODO(you): code here to reproduce the problem
```

kalyanvasudev, 2020-07-07:
Check if your base is running properly. This happens when either base is being detected or the base-related ros nodes crashed for some unknown reason.

Also, in the second terminal, the correct way to load the the necessary environment variables for the Pyrobot is `load_pyrobot_env` as mentioned here in the install script and in the tutorials - https://github.com/facebookresearch/pyrobot/blob/master/install_pyrobot.sh#L112

kalyanvasudev, 2020-07-07:
Feel free to re-open the issue if it persists.

closed 2020-07-07 by kalyanvasudev