facebookresearch/pyrobot #97: robot.base.get_state('odom') always returns [0,0,0]

fujin678, 2020-07-06:
## Steps to reproduce

  1. _____get current pose (robot.base.get_state('odom')),  which is [0,0,0]
  2. _____move locobot by running:
python base_position_control.py --base_planner none --base_controller ilqr --smooth  --close_loop --relative_position   0.2,0.00000001,0.0  --botname locobot
  3. ______get current pose again (robot.base.get_state('odom')),  which is still [0,0,0]

## Observed Results

  * What happened?  This could be a description, log output, etc.

## Expected Results

  * What did you expect to happen?

## Relevant Code

```
// TODO(you): code here to reproduce the problem
```

fujin678, 2020-07-06:
looks like the relative positions i used (0.2,0.00000001,0.0) are too small.   I just tried (1,0,0). it works as expected

kalyanvasudev, 2020-07-07:
Closing the issue since it is working

closed 2020-07-07 by kalyanvasudev