facebookresearch/pyrobot #98: yaw returned from  robot.base.get_state("vslam")  is not correct

fujin678, 2020-07-07:
## Steps to reproduce

  1.  run:
roslaunch locobot_control main.launch use_arm:=true use_base:=true use_camera:=true use_rviz:=false use_vslam:=true

  2.  in a new terminal, run
python base_position_control.py --base_planner none --base_controller ilqr --smooth  --close_loop --relative_position   1.3,1.1,0.8  --botname locobot

when it is finished, run 
r.base.get_state("vslam")
(array([1.34875655, 1.07990539, 0.01539037]), array([[ 0.69127667, -0.72255811,  0.00680753],
       [ 0.72235108,  0.6912596 ,  0.01921161],
       [-0.01858727, -0.00836311,  0.99979226]]), array([[ 0.69127667, -0.72255811,  0.00680753,  1.34875655],
       [ 0.72235108,  0.6912596 ,  0.01921161,  1.07990539],
       [-0.01858727, -0.00836311,  0.99979226,  0.01539037],
       [ 0.        ,  0.        ,  0.        ,  1.        ]]))

  3. ___

## Observed Results
yaw from robot.base.get_state("vslam") is  0.01539037,  the expected value is 0.8

For reference, robot.base.get_state("odom") shows correct results

r.base.get_state("odom")
[1.284969687461853, 1.0872259140014648, 0.8089600801467896]
>>> 

  * What happened?  This could be a description, log output, etc.

## Expected Results

  * What did you expect to happen?

## Relevant Code

```
// TODO(you): code here to reproduce the problem
```

kalyanvasudev, 2020-07-07:
'vslam'(ORB-SLAM) is known to very noisy on the locobot. Particularly in the context of yaw. To improve performance, move the robot around a bit and let it built a bigger libary of feature points to track for loop closure in SLAM. This should improve the performance.

kalyanvasudev, 2020-07-07:
If not, We have a version were we can do SLAM with Azure Kinect using R-TAB SLAM. This is a step up in SLAM performance when compared to ORB-SLAM with Realsense.

fujin678, 2020-07-07:
Hi kalyanvasudev,
thanks for your advice.

I found this is very interesting:
yaw from robot.base.get_state("vslam") is not correct.  But   yaw from  bot.base.base_state.vslam.base_pose_xyyaw looks pretty good.   So i guess my questions are,
- should  robot.base.get_state("vslam") and bot.base.base_state.vslam.base_pose_xyyaw return the same results?
- does bot.base.base_state.vslam.base_pose_xyyaw use vslam?

thanks,
Jeffrey

kalyanvasudev, 2020-07-07:
As mentioned here - https://github.com/facebookresearch/pyrobot/blob/89b25c039d504deeae35f28236f1c66554f12ea6/robots/LoCoBot/locobot_navigation/orb_slam2_ros/scripts/orb_slam2_ros/vslam.py#L154

`vslam.base_pose_xyyaw` would return (x,y, yaw) whereas `vlam.base_pose` would return translation vector, rotation matrix and the 4*4 transformation matrix.  

To be consistent with ` robot.base.get_state("odom")`, I will also change the robot.base.get_state("vslam") to return `vsalm.base_pose_xyyaw'

closed 2020-07-07 by kalyanvasudev