facebookresearch/pyrobot #99: rviz_locobot process has died

fujin678, 2020-07-14:
## Steps to reproduce

  1. run 👍 
roslaunch locobot_control main.launch use_arm:=true use_base:=true use_camera:=true **use_rviz:=true** use_vslam:=true

  2. _____
  3. _____

## Observed Results

[rviz_locobot_28759_2221210122411153159-20] **process has died** [pid 28938, exit code -11, cmd /opt/ros/kinetic/lib/rviz/rviz -d /home/locobot/low_cost_ws/src/pyrobot/robots/LoCoBot/locobot_moveit_config/launch/moveit.rviz __name:=rviz_locobot_28759_2221210122411153159 __log:=/home/locobot/.ros/log/ce06a5fe-c5e8-11ea-9916-d8f2cae6f041/rviz_locobot_28759_2221210122411153159-20.log].
log file: /home/locobot/.ros/log/ce06a5fe-c5e8-11ea-9916-d8f2cae6f041/rviz_locobot_28759_2221210122411153159-20*.log

actually, no rviz log file generated.

## Expected Results

  * What did you expect to happen?

## Relevant Code

```
// TODO(you): code here to reproduce the problem
```

swinterbotix, 2020-07-15:
If you ran that command over SSH without using the '-X' flag (so you did 'ssh locobot@locobot.local' instead of 'ssh -X locobot@locobot.local), that could happen.

fujin678, 2020-07-15:
found this is related to VNC.   so close it

closed 2020-07-15 by fujin678

fujin678, 2020-07-15:
i see, thanks