facebookresearch/pyrobot #100: when building map with orb-slam,  i got blank map

fujin678, 2020-07-15:
## Steps to reproduce

  1. _____roslaunch locobot_control main.launch use_arm:=true use_base:=true use_camera:=true  use_vslam:=true  use_rviz:=false
  2. _____drive robot to move  forward 1m.    I made sure camera can see some obstacles (eg, desk) in 2 meters

         python base_position_control.py --base_planner none  --base_controller proportional   --smooth=False --close_loop --relative_position 1.0,0.0,0.0 --botname locobot --use_map=False

  3. _____rosrun map_server map_saver -f test

## Observed Results

  dumped test.pgm  only has value oxfe (besides headers)

## Expected Results

  * What did you expect to happen?

## Relevant Code

```
// TODO(you): code here to reproduce the problem
```

kalyanvasudev, 2020-07-16:
ORBSLAM doesn't publish out the map as expected by the map_sever library. 

If you wish to use map_server, you need to publish in the compatible format on your own.

You could use this script to do so - https://github.com/facebookresearch/pyrobot/blob/e3126f007ac18518c2a0f3c004569ad785bb525d/robots/LoCoBot/locobot_navigation/base_navigation/nodes/mapToGrid.py

closed 2020-07-16 by kalyanvasudev

fujin678, 2020-07-16:
Hi kalyanvasudev,
your recently modified mapTogrid runs successfully.   But even with it,  i still get blank map.   Below is my test procedure,

1.  roslaunch locobot_control main.launch use_arm:=true use_base:=true use_camera:=true use_vslam:=true use_rviz:=false

2.  run "python mapToGrid.py".   Looks running OK.

3.  run "rosrun map_server map_saver -f test"

 the resulting "test.pgm" is blank.       Looks like map_server  does not receive the maps published by mapTOgrid