facebookresearch/pyrobot #104: [move_base-17] process has died

fujin678, 2020-07-20:
## Steps to reproduce

  1. _____run

roslaunch locobot_control main.launch use_arm:=true use_base:=true use_camera:=true use_rviz:=false   use_vslam:=true

  2. _____in a separate terminal, run

python mapToGrid.py

  3. _____ in the 3rd window, run

python base_position_control_with_map.py  --relative_position 2.0,0.0,0.0 --botname locobot

## Observed Results
after command "python base_position_control_with_map.py  --relative_position 2.0,0.0,0.0 --botname locobot",  locobot starts to move, but below message shows up after a few planning steps,

[move_base-17] process has died [pid 7116, exit code -11, cmd /opt/ros/kinetic/lib/move_base/move_base cmd_vel:=navigation_velocity_smoother/raw_cmd_vel odom:=odom scan:=scan __name:=move_base __log:=/home/locobot/.ros/log/fcfe4f6c-ca9b-11ea-9f12-d8f2cae6f041/move_base-17.log].
log file: /home/locobot/.ros/log/fcfe4f6c-ca9b-11ea-9f12-d8f2cae6f041/move_base-17*.log
  * What happened?  This could be a description, log output, etc.

## Expected Results

  * What did you expect to happen?

## Relevant Code

```
// TODO(you): code here to reproduce the problem
```

kalyanvasudev, 2020-07-29:
Hi, Could you please attach the full move_base log here ?

kalyanvasudev, 2020-08-28:
Closing the issue due to inactivity.

closed 2020-08-28 by kalyanvasudev