facebookresearch/pyrobot #108: Depth Image Demo

swinterbotix, 2020-07-23:
I've done the full install using Python 2 on an Interbotix Locobot on the latest commit to master as of this date

I noticed that while I am able to get Depth images from the camera_image.py script in the examples/locobot/navigation directory, the camera_control.py script just gives me a black window - I'm not seeing any grayish or white pixels - just pure blackness. Not exactly sure what the difference between these two scripts are that in the camera_image.py script, depth capture works and in camera_control.py it doesn't - but though it worth pointing out.

As an FYI, the RGB capture works just fine in both example scripts.

fujin678, 2020-08-19:
try to 👍 
   - #cv2.imshow("Depth", depth)
    + cv2.imshow("Depth", actual_depth_values)

kalyanvasudev, 2020-09-17:
I hope @fujin678 , response answers the question? I

closed 2020-09-17 by kalyanvasudev