facebookresearch/pyrobot #112: https://pyrobot.org/ is not accessible

fujin678, 2020-08-19:
## Steps to reproduce

  1. on chrome browser, visit https://pyrobot.org/
  2. _____
  3. _____

## Observed Results

This site can’t be reachedwww.pyrobot.org’s server IP address could not be found.
Try:

Checking the connection
Checking the proxy, firewall, and DNS configuration
Running Windows Network Diagnostics
ERR_NAME_NOT_RESOLVED

## Expected Results

  * What did you expect to happen?

## Relevant Code

```
// TODO(you): code here to reproduce the problem
```

yuanzhedong, 2020-08-27:
same here, can't access pyrobot.org

kalyanvasudev, 2020-08-27:
Hi all, we are working on it. We are hoping to have the website back up  in the next 2-3 days. Thanks!

kalyanvasudev, 2020-08-28:
The website is back up now. Please re-open the issue if the problems persist.

closed 2020-08-28 by kalyanvasudev