facebookresearch/pyrobot #117: No module named 'pyrobot'?

YangXJ95, 2020-09-14:
Hello I have just finished the installation of PyRobot and LoCoBot dependencies, and started to follow the "GET STARTED" tutorial, then by running the sample code I found that there is actually no package named pyrobot installed, also I tried "pip install pyrobot" and the pacakge seems to be out of date and import still failed...

kalyanvasudev, 2020-09-15:
Hi, What is the exact  shell command you ran for the installation? If using Python3, are the directories `low_cost_ws`, `pyrobot_catkin_ws`, `pyenv_pyrobot_python3` created by the install script?

YangXJ95, 2020-09-15:
Hello, it's `./locobot_install_all.sh -t full -p 3 -l interbotix`, yes these three directories were created.

kalyanvasudev, 2020-09-16:
How are you running the scripts? Is it inside a virtual environment sourced with the command `load_pyrobot_env`?

https://pyrobot.org/docs/navigation

YangXJ95, 2020-09-17:
I just realized I might just skip the `load_pyrobot_env` step all the time...very sorry for the unnecessary bother

closed 2020-09-17 by YangXJ95