facebookresearch/pyrobot #123: locobot: Source frame elbow_link does not exist

MithellScott, 2020-09-22:
## Steps to reproduce

  1. Set install script to install workspaces into a workspace dir and environments into an env dir
  2. source py3_ws, pyrobot_ws and py3_env: roslaunch locobot_control main.launch use_sim:=true use_base:=true
  3. source py3_ws, pyrobot_ws and py3_env: in python3 terminal from pyrobot import Robot; bot = Robot('locobot')

## Observed Results
tf2.TransformException: canTransform: source elbow link does not exist
## Expected Results
The expected result is that the locobot will initialize and allow control of the gazebo rendering from the console.
This is essentially the navigation exercise from the pyrobot/locobot tutorials.
## Relevant Code
Roslaunch output:
[image]

[image]

Install Scripts:
[locobot_install_all.txt](https://github.com/facebookresearch/pyrobot/files/5263113/locobot_install_all.txt)
[install_pyrobot.txt](https://github.com/facebookresearch/pyrobot/files/5263115/install_pyrobot.txt)

```

kalyanvasudev, 2020-09-23:
hi, could you please elaborate the commands you ran in the 1st terminal?

kalyanvasudev, 2020-09-23:
>     2\. source py3_ws, pyrobot_ws and py3_env: roslaunch locobot_control main.launch use_sim:=true use_base:=true

Ah I see. You are doing this step wrong. The command should not run inside the virtual environment. 

And for step 3, use the alias created by the install script `load_pyrobot_env` to load the appropriate env.

kalyanvasudev, 2020-09-23:
Please feel free to re-open the issue if the problem persists.

closed 2020-09-23 by kalyanvasudev

MithellScott, 2020-09-23:
I fixed the problem by reinstalling the sim only version. Also when following your instructions above I was not able to fix the issue with the full install(I am running ubuntu 18 and python3).  The error is different, I think from online it has something to do with python file bindings
[image]