facebookresearch/pyrobot #124: Locobot Navigation tutorial

MithellScott, 2020-09-23:
## Steps to reproduce
1.) Use Ubuntu 18.04 and python3
2.) Run the full installation for Locobot
    -`locobot_install_all -t full -p 3 -l interbotix`
3.) Follow the navigation tutorial.
    - In one console
           `source low_cost_ws/devel/setup.bash`
            `roslaunch locobot_control main.launch use_base:=true use_sim:=true` 
    - In another console 
            `source pyenv_pyrobot_python3/bin/activate` 
            `source pyrobot_catkin_ws/devel/setup.bash` (using alias)
             ` rosrun locobot_riss CircleDr.py` 

**locobot_riss is a package I created with nothing more than the CircleDr.py file. This package was not created with dependencies.
## Observed Results

[image]

## Expected Results
This should cause the Locobot to drive in a large circle. This problem does not occur when installing the simulation only. From what I have read online this is an issue with the python version binding to a *.so file. That is far beyond me so I  hope that someone here will have a more informed opinion.

## Relevant Code
[CircleDr.txt](https://github.com/facebookresearch/pyrobot/files/5269445/CircleDr.txt)

kalyanvasudev, 2020-09-23:
You seem to be working with a non-standard code supported by us?  How is your `pyro_ws` compiled? 

Can you confirm that the examples mentioned in tutorial are working?

MithellScott, 2020-09-23:
I have not tried any examples other than the navigation. I changed the names of some workspaces in the install script but the configuration is the same. I tried this with the normal facebookresearch install script as well and got the same issue.

kalyanvasudev, 2020-09-24:
Please try the examples and let me know. Also, rosrun is not the standard way of running any pyrobot scripts unless properly compiled with the right paths.

MithellScott, 2020-09-26:
Good news I got the examples to work and soon after my own scripts. On multiple occasions I have run into an issue no module named netifaces which pops up in rosgraph.py. Im not sure if you have seen this but I fixed it by re-installing. Also While I have spent the last few days periodically running my script to make sure that the issue is truly resolved. I have seen the above issue pop up now and again, but I must be screwing up what I source when running things. Just to be totally clear I run the launch script in low_cost_ws, then I run my script in low_cost + pyrobot_catkin_ws + pyenv_pyrobot_python3(respectively). The navigation tutorial online only says to source the py3 env but I get a missing pyrobot-bridge w/out low_cost_ws and w/out pyrobot_catkin_ws I get a the above error (of course without either I get a missing locobot_control package error). Also running the launch with pyrobot_catkin_ws can lead to these errors.

closed 2020-09-26 by MithellScott