facebookresearch/pyrobot #125: Problem "Neither use_base, nor use_sim, is not set to True"

YangXJ95, 2020-09-23:
Hello, I ran into the problem while trying to get the robot running:

1. turn locobot on(base status light green, U2D2 light red, NUC blue)

1. 

ssh with terminal header turned to: locobot@locobot:
in the terminal with locobot@locobot: run
roslaunch locobot_control main.launch use_base:=true 
hear the beep of connecting to the base, and finally shows "You can start planning now!"
new terminal run "roscore"
new terminal cd to python code directory, run "load_pyrobot_env" and have the header start with (pyenv_pyrobot_python3, then run "python3 XXX.py" (code from here: https://pyrobot.org/docs/navigation)
and the errors and warnings I get from terminals:
1. for terminal which runs roslaunch, before showing "You can start planning now!" there is one error:
[rviz_locobot_17524_4602792288901938809-14] process has died [pid 17644, exit code -6, cmd /opt/ros/kinetic/lib/rviz/rviz -d /home/locobot/low_cost_ws/src/pyrobot/robots/LoCoBot/locobot_moveit_config/launch/moveit.rviz __name:=rviz_locobot_17524_4602792288901938809 __log:=/home/locobot/.ros/log/fc1ef444-fe07-11ea-860e-24418c89d43d/rviz_locobot_17524_4602792288901938809-14.log].

2.for terminal which runs python code, I got three warnings and then it shut down:
[WARN] [1600882101.574117]: Neither use_arm, nor use_sim, is not set to True when the LoCoBot driver is launched.You may not be able to command the arm correctly using PyRobot!!!
[WARN] [1600882101.972273]: Neither use_base, nor use_sim, is not set to True when the LoCoBot driver is launched. You may not be able to command the base correctly using PyRobot!
[WARN] [1600882101.974842]: Neither use_camera, nor use_sim, is not set to True when the LoCoBot driver is launched.You may not be able to command the camera correctly using PyRobot!!!

I am confused by "Neither...nor..., is not..., may not be able to command..." and wondering which part went wrong with connection failed.

YangXJ95, 2020-09-24:
Also can I ask is there any other method that I can take to command the movement of the robot base?

kalyanvasudev, 2020-09-24:
You dont need to do step 5.

After step 4, could you open a new terminal and run `rostopic echo /odom` and see if the odommetry message is being published?

kalyanvasudev, 2020-09-24:
Also, could you confirm if step 6 is working in gazebo? You can do this by ssh -X and  by running the following command in step 3 , `roslaunch locobot_control main.launch use_base:=true use_sim:=true`

YangXJ95, 2020-09-24:
By running `rostopic echo /odom` it gives out `ERROR: Unable to communicate with master!`...

YangXJ95, 2020-09-24:
`roslaunch locobot_control main.launch use_base:=true use_sim:=true` gives out the following and it didn't work:
` * /move_base/global_costmap/obstacle_layer/z_voxels: 2
 * /move_base/global_costmap/plugins: [{'type': 'costma...
 * /move_base/global_costmap/publish_frequency: 0.5
 * /move_base/global_costmap/resolution: 0.2
 * /move_base/global_costmap/robot_base_frame: base_footprint
 * /move_base/global_costmap/robot_radius: 0.2
 * /move_base/global_costmap/rolling_window: True
 * /move_base/global_costmap/static_layer/enabled: True
 * /move_base/global_costmap/static_map: True
 * /move_base/global_costmap/transform_tolerance: 0.5
 * /move_base/global_costmap/update_frequency: 1.0
 * /move_base/global_costmap/width: 100.0
 * /move_base/local_costmap/always_send_full_costmap: True
 * /move_base/local_costmap/global_frame: odom
 * /move_base/local_costmap/height: 4.0
 * /move_base/local_costmap/inflation_layer/cost_scaling_factor: 5.0
 * /move_base/local_costmap/inflation_layer/enabled: True
 * /move_base/local_costmap/inflation_layer/inflation_radius: 0.5
 * /move_base/local_costmap/map_type: costmap
 * /move_base/local_costmap/max_obstacle_height: 0.6
 * /move_base/local_costmap/obstacle_layer/bump/clearing: False
 * /move_base/local_costmap/obstacle_layer/bump/data_type: PointCloud2
 * /move_base/local_costmap/obstacle_layer/bump/marking: True
 * /move_base/local_costmap/obstacle_layer/bump/max_obstacle_height: 0.15
 * /move_base/local_costmap/obstacle_layer/bump/min_obstacle_height: 0.0
 * /move_base/local_costmap/obstacle_layer/bump/topic: mobile_base/senso...
 * /move_base/local_costmap/obstacle_layer/combination_method: 1
 * /move_base/local_costmap/obstacle_layer/enabled: True
 * /move_base/local_costmap/obstacle_layer/mark_threshold: 0
 * /move_base/local_costmap/obstacle_layer/max_obstacle_height: 0.6
 * /move_base/local_costmap/obstacle_layer/observation_sources: scan bump
 * /move_base/local_costmap/obstacle_layer/obstacle_range: 2.5
 * /move_base/local_costmap/obstacle_layer/origin_z: 0.0
 * /move_base/local_costmap/obstacle_layer/publish_voxel_map: False
 * /move_base/local_costmap/obstacle_layer/raytrace_range: 3.0
 * /move_base/local_costmap/obstacle_layer/scan/clearing: True
 * /move_base/local_costmap/obstacle_layer/scan/data_type: LaserScan
 * /move_base/local_costmap/obstacle_layer/scan/marking: True
 * /move_base/local_costmap/obstacle_layer/scan/max_obstacle_height: 0.35
 * /move_base/local_costmap/obstacle_layer/scan/min_obstacle_height: 0.25
 * /move_base/local_costmap/obstacle_layer/scan/topic: scan
 * /move_base/local_costmap/obstacle_layer/track_unknown_space: True
 * /move_base/local_costmap/obstacle_layer/unknown_threshold: 15
 * /move_base/local_costmap/obstacle_layer/z_resolution: 0.2
 * /move_base/local_costmap/obstacle_layer/z_voxels: 2
 * /move_base/local_costmap/plugins: [{'type': 'costma...
 * /move_base/local_costmap/publish_frequency: 2.0
 * /move_base/local_costmap/resolution: 0.01
 * /move_base/local_costmap/robot_base_frame: base_footprint
 * /move_base/local_costmap/robot_radius: 0.2
 * /move_base/local_costmap/rolling_window: True
 * /move_base/local_costmap/static_layer/enabled: True
 * /move_base/local_costmap/static_map: False
 * /move_base/local_costmap/transform_tolerance: 0.5
 * /move_base/local_costmap/update_frequency: 5.0
 * /move_base/local_costmap/width: 4.0
 * /move_base/oscillation_distance: 0.2
 * /move_base/oscillation_timeout: 10.0
 * /move_base/planner_frequency: 1.0
 * /move_base/planner_patience: 1.0
 * /move_base/recovery_behavior_enabled: False
 * /move_base/shutdown_costmaps: False
 * /move_group/allow_trajectory_execution: True
 * /move_group/arm/default_planner_config: ESTkConfigDefault
 * /move_group/arm/longest_valid_segment_fraction: 0.005
 * /move_group/arm/planner_configs: ['SBLkConfigDefau...
 * /move_group/arm/projection_evaluator: joints(joint_1,jo...
 * /move_group/capabilities: 
 * /move_group/controller_list: [{'default': True...
[... 342 lines cut ...]
Loading 'move_group/MoveGroupPlanService'...
Loading 'move_group/MoveGroupQueryPlannersService'...
Loading 'move_group/MoveGroupStateValidationService'...
[ INFO] [1601006520.697070693, 12.404000000]: 

********************************************************
* MoveGroup using: 
*     - ApplyPlanningSceneService
*     - ClearOctomapService
*     - CartesianPathService
*     - ExecuteTrajectoryAction
*     - GetPlanningSceneService
*     - KinematicsService
*     - MoveAction
*     - PickPlaceAction
*     - MotionPlanService
*     - QueryPlannersService
*     - StateValidationService
********************************************************

[ INFO] [1601006520.697143266, 12.404000000]: MoveGroup context using planning plugin ompl_interface/OMPLPlanner
[ INFO] [1601006520.697172441, 12.404000000]: MoveGroup context initialization complete

You can start planning now!
`

YangXJ95, 2020-09-30:
@kalyanvasudev Hello, I think it might be the problem of the hotspot connection, and I directly accessed Intel NUC with HDMI and keyboard. I logged into the Locobot's Ubuntu system and run the `roslaunch locobot_control main.launch use_base:=true`, and also the command `rostopic echo /odom` and it is feeding back odom info. Though running the python code I only got `ImportError: No module named 'pyrobot'`, and there is no `pyrobot` package in neither `pip list` nor `pip3 list`. I am confused, for NUC to command the base movement there is no need to import `pyrobot`? If so, how can I control the base movement directly from NUC's Ubuntu system?

YangXJ95, 2020-09-30:
It's the Locobot NUC's Pyrobot package out-of-date which results in the malfunction.

closed 2020-09-30 by YangXJ95