facebookresearch/pyrobot #139: get_current_pcd from camera.py returns empty list

Jendker, 2020-11-25:
## Steps to reproduce

  1. install pyrobot with either python2 or python3
  2. run `roslaunch locobot_control main.launch use_arm:=true use_camera:=true use_base:=true use_rviz:=false`
  3. `get_current_pcd` function returns empty list if executed in script

## Observed Results

  * I have reinstalled the robot because of problem with the software and now the `get_current_pcd` (https://github.com/facebookresearch/pyrobot/blob/master/src/pyrobot/locobot/camera.py#L68) and now it returns empty array.

## Expected Results

  * Before the software reinstall the pointcloud was returned correctly.

## Relevant Code

```
from pyrobot import Robot

bot = Robot('locobot')
print('pcl', bot.camera.get_current_pcd(True))  # or False, does not matter
```

## Observation
When I followed the output values, it seems that processing stops at line https://github.com/facebookresearch/pyrobot/blob/4b3f0135f2610646e89840329cd1a2db3db5ee63/src/pyrobot/locobot/camera.py#L446 where I get all indices set to False.

Jendker, 2020-11-27:
I tried following the steps from here https://pyrobot.org/docs/faq step 4, the firmware install and driver install did not help.
I am not sure though how to check if ORB-SLAM2 is running and tracking, maybe that is the issue? Any hint how to check it?

Jekyll1021, 2020-11-27:
I noticed the line you quoted (Line 446 in `camera.py`) stopped because depth image doesn't fall in expected range.
I suspect the problem might be even earlier in the process - maybe the depth input itself is invalid (all NaN, for example.)
Could you try printing `depth` and attach what you get here?

Jendker, 2020-11-27:
Thanks for answer! I'll do it soon, but first wanted to say that all `pytest test_camera.py -v` tests passed.

Jekyll1021, 2020-11-27:
I'm assuming you are using orb_slam on a real robot (it doesn't work in sim anyways). `test_camera.py` runs test in sim, so I think it's worth checking depth works properly in real as a starting point.

Jendker, 2020-11-27:
I get such values for `depth` and queried max and min values:
'depth' [0.    0.    0.    ... 0.001 0.001 0.001]
'max'  0.035)
'min' 0.0

Jekyll1021, 2020-11-27:
The unit is in meters. Does that align with what you expected? Assuming you have camera mounted on the robot and the robot on the floor, this seems abnormally small. 
I suggest visualizing depth/point cloud in rviz to see if the input makes sense.

Jendker, 2020-11-27:
The visualization in Rviz looks fine. I have tried to swap the camera with another Intel RealSense D435, but when running roslaunch got errors that no frames are arriving. Is there any way to set up the new camera quickly to check for hardware issue? If that would be advisable.

Jekyll1021, 2020-11-27:
You can change the camera configs and slam configs accordingly: https://github.com/facebookresearch/pyrobot/blob/4b3f0135f2610646e89840329cd1a2db3db5ee63/src/pyrobot/cfg/locobot_config.py#L39

https://github.com/facebookresearch/pyrobot/blob/4b3f0135f2610646e89840329cd1a2db3db5ee63/src/pyrobot/cfg/locobot_config.py#L127

Jendker, 2020-11-27:
I have run the calibration of camera, but the values are still way too small while Rviz visualization presents everything fine. With yet another camera I have the exact issue.

Jendker, 2020-11-27:
Now I have found something in the output of `roslaunch`:

```
Traceback (most recent call last):
  File "/home/locobot/low_cost_ws/src/pyrobot/robots/LoCoBot/locobot_control/nodes/pointrobot3factor_ros_server.py", line 7, in <module>
    from gtsam import *
ImportError: No module named gtsam
```
not sure what causes it though.

Jekyll1021, 2020-11-27:
> I have run the calibration of camera, but the values are still way too small while Rviz visualization presents everything fine. With yet another camera I have the exact issue. I changed the USB port, still get empty array.
> 
> I am getting closer and closer to the idea of cloning another locobot hard drive onto this locobot.

Also: could you check what the `depth/image_raw` rostopic returns? Curious to see if that aligns with what you get for depth * 1000

Jendker, 2020-11-27:
There is no `depth/image_raw` but `depth/image_rect_raw` and there I get integer values 0-255.

Jekyll1021, 2020-11-27:
> Now I have found something in the output of `roslaunch locobot_control main.launch use_arm:=true use_camera:=true use_base:=true use_rviz:=false`:
> 
> ```
> Traceback (most recent call last):
>   File "/home/locobot/low_cost_ws/src/pyrobot/robots/LoCoBot/locobot_control/nodes/pointrobot3factor_ros_server.py", line 7, in <module>
>     from gtsam import *
> ImportError: No module named gtsam
> ```
> 
> not sure what causes it, or why is it missing though.

Could you disable https://github.com/facebookresearch/pyrobot/blob/4b3f0135f2610646e89840329cd1a2db3db5ee63/robots/LoCoBot/locobot_control/launch/main.launch#L84 
this line for now? `pointrobot3factor_ros_server.py` is specifically built for using gpmp controller for base planning, as far as I'm aware.

Jekyll1021, 2020-11-27:
> There is no `depth/image_raw` but `depth/image_rect_raw` and there I get integer values 0-255.

Sorry, should be `/camera/aligned_depth_to_color/image_raw`. Also 0-255 looks skeptical, should be in millimeters.

Jendker, 2020-11-27:
> Could you disable
> 
> https://github.com/facebookresearch/pyrobot/blob/4b3f0135f2610646e89840329cd1a2db3db5ee63/robots/LoCoBot/locobot_control/launch/main.launch#L84
> 
> 
> this line for now?

This does not prevent the error.

> Sorry, should be /camera/aligned_depth_to_color/image_raw. Also 0-255 looks skeptical, should be in millimeters.

I get the same output, integers 0-255.

Jekyll1021, 2020-11-27:
> > Could you disable
> > https://github.com/facebookresearch/pyrobot/blob/4b3f0135f2610646e89840329cd1a2db3db5ee63/robots/LoCoBot/locobot_control/launch/main.launch#L84
> > 
> > this line for now?
> 
> This does not prevent the error.

Did you comment out the following line also? This is the only place `pointrobot3factor_ros_server.py` get called.

> > Sorry, should be /camera/aligned_depth_to_color/image_raw. Also 0-255 looks skeptical, should be in millimeters.
> 
> I get the same output, integers 0-255.

Ah. I think that's the issue. The depth image looks normalized but it really shouldn't. Let me double check with my colleague who has a locobot at hand - I don't have one at this moment.

Jendker, 2020-11-27:
> Did you comment out the following line also? This is the only place `pointrobot3factor_ros_server.py` get called.

You are absolutely right, I was trying to comment it out with '#'... I am getting tired.

> Ah. I think that's the issue (not 100% sure tho). The depth image looks normalized but it really shouldn't. Let me double check with my colleague who has a locobot at hand - I don't have one at this moment.

Yes, that could definitely cause some problems with the threshold. Please let me know what you find out! And thank you very much for your time!

Jendker, 2020-11-30:
I wanted to kindly ask if there is any new input on this issue?

Jekyll1021, 2020-11-30:
We were on vacation last week (for Thanksgiving). We'll take a look after get back to work.

Jendker, 2020-11-30:
Thank you very much for the help even this Thursday then!

Jekyll1021, 2020-11-30:
I checked with my colleague @Dhiraj100892 and he fixed the problem in this PR: https://github.com/facebookresearch/pyrobot/pull/135/files

Could you pull the latest `Develop` branch and see if the problem has fixed??

Jendker, 2020-12-01:
Unfortunately I still get the empty array. When I run the same check as in this comment: https://github.com/facebookresearch/pyrobot/issues/139#issuecomment-734630756 I receive the following output

'depth' [0.001 0.001 0.001 ... 0.001 0.001 0.001]
'max', 0.002
'min', 0.0

I am not sure if that helps, but these values are comparable to those if I run the code on `master` branch.

Dhiraj100892, 2020-12-01:
@Jendker After pulling the `Develop` branch did you reinstall PyRobot ?
Basically `load_pyrobot_env` and run `pip install .` from Pyrobot root directory

Jendker, 2020-12-01:
Yes, I forgot to reinstall PyRobot. I got used to using packages with the editable option. It works fine now, thank you for the quick fix!

closed 2020-12-01 by Jendker