facebookresearch/pyrobot #144: Locobot Python 3 install on Ubuntu 16.04 Issue

swiz23, 2021-02-02:
When running `./locobot_install_all.sh -t full -p 3 -l interbotix`, the install process exits early with the message below...

[image]

I believe this issue happens when running [this line](https://github.com/facebookresearch/pyrobot/blob/4b3f0135f2610646e89840329cd1a2db3db5ee63/robots/LoCoBot/install/locobot_install_all.sh#L380) of code (`sudo pip install catkin_pkg pyyaml empy rospkg`).

That is because in [this line](https://github.com/facebookresearch/pyrobot/blob/4b3f0135f2610646e89840329cd1a2db3db5ee63/robots/LoCoBot/install/locobot_install_all.sh#L110) of code above (`sudo pip install --upgrade pip`), you are upgrading pip, which installs the latest version (21.0). Unfortunately, this version drops support for Python2 packages (see [here](https://pip.pypa.io/en/stable/news/#id4)), which is my guess for why these error messages are showing up. 

Not sure how to fix this... either don't upgrade pip, or if you do upgrade pip, install python3-pip and then use pip3 instead of pip?

Jekyll1021, 2021-02-02:
Could you tell me how exactly to replicate the bug? I just ran an install on a clean machine and it worked fine.

swiz23, 2021-02-02:
I have a client who encountered this issue. I then tested it myself on a clean machine and replicated the issue myself. I used a clean install of Linux 16.04.7 on a NUC8i3BEH. After logging in for the first time, i just ran _sudo apt update_ and _sudo apt upgrade_. Then I typed `./locobot_install_all.sh -t full -p 3 -l interbotix` in a terminal (after curling the install script), and that was it.

Is it possible that you grabbed the install script from the development branch or did not update the computer? I noticed that the latest version of 'pip' dropped support for Python 2 on January 23rd.

Jekyll1021, 2021-02-02:
Thanks for the detailed instruction - it very much likely to be the latter. Let me try it and get back to you.

Jekyll1021, 2021-02-03:
Closing the issue - PR #145 should address this. Thank you swiz23!

closed 2021-02-03 by Jekyll1021

swiz23, 2021-02-03:
Awesome. I saw that some CI checks didn't pass though. How will that effect the installation procedure?

Jekyll1021, 2021-02-03:
> Awesome. I saw that some CI checks didn't pass though. How will that affect the installation procedure?

It doesn't - if you check the commit, the only line changed is to specify a pip version, as you suggested. The CI didn't pass because I manually killed the job and used CI cluster as testing machine (the testing procedure installed from /project folder, instead of installation script, which is why we overlooked this issue).

swiz23, 2021-02-03:
Ok great!

swiz23, 2021-02-04:
How do you go about downgrading pip? Now that my system has pip v21, I can't even use the 'pip' command to get a downgraded version. Even just running 'pip' in the terminal results in the error message above.

Jekyll1021, 2021-02-04:
Try `sudo apt-get remove python-pip`?

swiz23, 2021-02-04:
Nope, that doesn't work. I'm able to remove python-pip (which seems to only remove v8.1.1 - the default), but running the second commands results in an error. Also, the `apt-cache policy` command shows v8.1.1 as the only available package.

I think where python-pip stuff is installed and where 'pip' stuff is installed are in two separate locations. If there was just a way to wipe out the pip version 21 packages from the system, I image the upgrade pip command as shown in the install script will work - but not sure how to do this.

Jekyll1021, 2021-02-04:
Try `sudo pip install --force-reinstall pip==20.3.4`?

swiz23, 2021-02-04:
Yup - tried that too, but nothing doing

Jekyll1021, 2021-02-04:
The ugly way might work - delete the pip package installed at /usr/local/lib/python2.7/dist-packages/pip, then run sudo apt-get install --reinstall python2.7 (which in theory should get the default pip tied to py2.7 back)

swiz23, 2021-02-04:
So I deleted the package at `/usr/local/lib/python2.7/dist-packages/pip`. I also deleted `/usr/local/bin/pip` (at least I think that was the location). Afterwards, when I ran `pip -V` in the terminal, it pointed to my original python-pip installation (v8.1.1) located at `/usr/lib/python2.7/dist-packages` (I reinstalled it before doing this whole thing).

Then when I ran the installation script again, it updated to pip 20.3 correctly.

So it doesn't look like there was any need to run `sudo apt-get install --reinstall python2.7` at all. Also, it looks like the 'apt' version of pip is installed in /usr/lib/... vs the pip version of pip is installed in /usr/local/lib/...