facebookresearch/pyrobot #151: collect_calibration_data crashes (Ubuntu 18.04, Python 2/3)

dogoepp, 2021-03-30:
Hi,

We just got our LoCoBot from Trossen Robotics and attempted to install and get running with Pyrobot. It had first not worked well with Ubuntu 16.04, which is old anyways so we tried again on Ubuntu 18.04. Now we are stuck with the issue below. Not being familiar with the way pyrobot is built, I would really appreciate some help on sorting out what is not working here.

## Steps to reproduce

  1. install pyrobot following the [official script](https://github.com/facebookresearch/pyrobot/blob/master/robots/LoCoBot/install/locobot_install_all.sh) on a fresh Ubuntu 18.04 (tried both python 3 and python 2)
  2. `roslaunch locobot_control main.launch use_arm:=true use_base:=true use_camera:=true use_rviz:=false`
  3. `roslaunch locobot_calibration ar_track_alvar_calibration.launch`
  4. source the virtual environment created during the installation phase
  4. Following the instructions on step 2 of [this tutorial](https://pyrobot.org/docs/calibration), run `collect_calibration_data.py `

## Observed Results

We systematically get this error message:

```
(pyenv_pyrobot_python2) locobot@locobot1:~/low_cost_ws/src/pyrobot/robots/LoCoBot/locobot_calibration$ python scripts/collect_calibration_data.py --data_dir tmp/calibration-data/v1/ --botname locobot
I0330 14:39:03.519524 140449073645376 collect_calibration_data.py:331] Storing data in directory: tmp/calibration-data/v1/
Traceback (most recent call last):
  File "scripts/collect_calibration_data.py", line 337, in <module>
    app.run(main)
  File "/home/locobot/pyenv_pyrobot_python2/local/lib/python2.7/site-packages/absl/app.py", line 300, in run
    _run_main(main, args)
  File "/home/locobot/pyenv_pyrobot_python2/local/lib/python2.7/site-packages/absl/app.py", line 251, in _run_main
    sys.exit(main(argv))
  File "scripts/collect_calibration_data.py", line 332, in main
    calibrator = CameraCalibration(FLAGS.botname)
  File "scripts/collect_calibration_data.py", line 90, in __init__
    self.bot = Robot(botname)
  File "/home/locobot/pyenv_pyrobot_python2/local/lib/python2.7/site-packages/pyrobot/core.py", line 138, in __init__
    mod = importlib.import_module(root_node + "base")
  File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
  File "/home/locobot/pyenv_pyrobot_python2/local/lib/python2.7/site-packages/pyrobot/locobot/base.py", line 17, in <module>
    from ca_msgs.msg import Bumper
ImportError: No module named ca_msgs.msg
```

## Expected Results

We expected that the arm would move so as to capture data for calibration of the Realsense camera.

dogoepp, 2021-03-31:
Hi Zisu,

Thanks for your reply and guidance. Is there an other way than reinstalling ubuntu to uninstall all the pyrobot software ? This way I could reproduce again the steps and be sure to do nothing more than the install script. Maybe it would be sufficient to delete all folders that were created in `/home/locobot`?

I also followed your instructions but to the same result as before, following the same steps I described above. Should I do something for the outputs of catkin_make to be available from within the virtual environment used for `collect_calibration_data.py`?

I will now try to  temporarily remove the lines related to `ca_msgs`, to see if it triggers other errors that could guide us in our quest.

dogoepp, 2021-03-31:
Now that I carefully removed the import of `ca_msgs` from ``, the following happens.

Firstly, the arm moved to one pose for the arm and a few positions for the camera and the camera too. Looks like the calibration will work. And then the node crashes with this error:

```
I0331 11:31:06.886723 139678700431168 collect_calibration_data.py:331] Storing data in directory: tmp/calibration-data/v1/
[INFO] [1617183078.991269]: Planning to pose [0.31, 0.21, -0.178, -0.623, -0.328]
Traceback (most recent call last):
  File "scripts/collect_calibration_data.py", line 337, in <module>
    app.run(main)
  File "/home/locobot/pyenv_pyrobot_python2/local/lib/python2.7/site-packages/absl/app.py", line 300, in run
    _run_main(main, args)
  File "/home/locobot/pyenv_pyrobot_python2/local/lib/python2.7/site-packages/absl/app.py", line 251, in _run_main
    sys.exit(main(argv))
  File "scripts/collect_calibration_data.py", line 333, in main
    calibrator.collect_data(FLAGS.data_dir)
  File "scripts/collect_calibration_data.py", line 248, in collect_data
    ar_pose.markers[0], "pose_img_{:d}".format(_)
AttributeError: 'AlvarMarker' object has no attribute 'pose_img_0'
[INFO] [1617183091.362764]: Stopping LoCoBot Base. Waiting for base thread to finish
```

I looked at the contents of the `/ar_pose_marker` topic. It seems that the marker recognition works. I cannot tell however if it is doing what is intended. The only unexpected thing I noticed is that the content of the `markers` field seems to be a list of ROS messages (with a full header). Maybe I missed something else...

Jekyll1021, 2021-03-31:
I looked into the protocol, and `pose_img_x` is indeed not included. I'm reaching out to who wrote the script - meanwhile, the easiest thing to do is probably just save translation and orientation, and optimize on 3d poses (which would lead to a lower accuracy then using all the corners).

dogoepp, 2021-04-14:
Thanks for this information. I'm deep in other issues around the robot now, but shall give it a try as soon as possible.

Jekyll1021, 2021-04-19:
Sure! Please reopen if you have further problem!

closed 2021-04-19 by Jekyll1021

seeeheng, 2021-04-29:
Hi @Jekyll1021 - am running a clean install of Pyrobot on a vanilla Ubuntu 18.04 as I encountered issues on Ubuntu 16.04 as well. I run into the same error where I get an error with pose_image_x, even **after** checking out the kinetic-devel branch.
 
```[INFO] [1619675832.418491]: Planning to pose [0.31, 0.21, -0.178, -0.623, -0.328]
Traceback (most recent call last):
  File "scripts/collect_calibration_data.py", line 337, in <module>
    app.run(main)
  File "/home/locobot3/pyenv_pyrobot_python3/lib/python3.6/site-packages/absl/app.py", line 300, in run
    _run_main(main, args)
  File "/home/locobot3/pyenv_pyrobot_python3/lib/python3.6/site-packages/absl/app.py", line 251, in _run_main
    sys.exit(main(argv))
  File "scripts/collect_calibration_data.py", line 333, in main
    calibrator.collect_data(FLAGS.data_dir)
  File "scripts/collect_calibration_data.py", line 248, in collect_data
    ar_pose.markers[0], "pose_img_{:d}".format(_)
AttributeError: 'AlvarMarker' object has no attribute 'pose_img_0'```

seeeheng, 2021-04-29:
I noticed the variable `ar_pose.markers[0]` no longer has an attribute `pose_img_0` as mentioned by the Traceback. I amended line 248 within `collect_calibration_data.py` to the current attribute, `pose`, and the script is no longer complaining. 

Here's a sample of my `ar_pose.markers[0]`.

```header: 
  seq: 0
  stamp: 
    secs: 1619676729
    nsecs: 822305243
  frame_id: "camera_link"
id: 0
confidence: 0
pose: 
  header: 
    seq: 0
    stamp: 
      secs: 0
      nsecs:         0
    frame_id: ''
  pose: 
    position: 
      x: 0.45269876415826776
      y: -0.1411925639236865
      z: 0.09951914404272649
    orientation: 
      x: -0.3123414296483348
      y: -0.3443004804322805
      z: 0.4842247030526258
      w: 0.7412330587965781

Jekyll1021, 2021-04-29:
> Hi @Jekyll1021 - am running a clean install of Pyrobot on a vanilla Ubuntu 18.04 as I encountered issues on Ubuntu 16.04 as well. I run into the same error where I get an error with pose_image_x, even **after** checking out the kinetic-devel branch.
> 
> ```
> Traceback (most recent call last):
>   File "scripts/collect_calibration_data.py", line 337, in <module>
>     app.run(main)
>   File "/home/locobot3/pyenv_pyrobot_python3/lib/python3.6/site-packages/absl/app.py", line 300, in run
>     _run_main(main, args)
>   File "/home/locobot3/pyenv_pyrobot_python3/lib/python3.6/site-packages/absl/app.py", line 251, in _run_main
>     sys.exit(main(argv))
>   File "scripts/collect_calibration_data.py", line 333, in main
>     calibrator.collect_data(FLAGS.data_dir)
>   File "scripts/collect_calibration_data.py", line 248, in collect_data
>     ar_pose.markers[0], "pose_img_{:d}".format(_)
> AttributeError: 'AlvarMarker' object has no attribute 'pose_img_0'
> ```

Did you `catkin_make` under root in folder `low_cost_ws` to recompile the workspace and reopen a new terminal for the change to take effect?

seeeheng, 2021-05-04:
Woops I realized my previous post is completely wrong and misleading, so I removed it. Turns out I forgot to build the ws after checking out kinetic-devel. Thanks @Jekyll1021 , it's working now.