facebookresearch/pyrobot #152: Bouncing effect seen

yoggieWX, 2021-04-15:
[image]
[image]

## Observed Results

  * Hi,

   We are using a locobot model in our project. We tried using the gripper to grip a brush and to draw a line backwards. We 
  noticed that during the drawing of the line, there was a bounce effect everytime it tried to draw back. When we did a plot of 
   the xyz, we noticed that the Z coordinates seemed to be heading for a huge offset.

  What we did was, we tried using simple commands below and observed still the same effect.
  - set_ee_pose, 
  - move_xyz, 
  - set_ee_pose_pitch_roll.

  I  have attached the plots for both the set_ee_pose and move_xyz for your reference.
  Could you advise us on what could be the possible problem?
  
   Also, could i check on the commands:
   robot.arm.set_ee_pose_pitch_roll(position, pitch=??, roll=0??,plan=True, wait=True)
   Does changing the plan=False and wait=False affect the movements?

## Expected Results

  * The robot arm is supposed to draw a straight line

Jekyll1021, 2021-04-15:
plan=True use moveit! planner and controller in the backend. If you have a trajectory that you would like to execute, I highly recommend you look into [move_group_interface.py](https://github.com/facebookresearch/pyrobot/blob/master/src/pyrobot/utils/move_group_interface.py) - directly commanding to waypoints is usually quite bumpy. 

Also, would you mind going through what units the plot are in and how the plot data is gathered?

yoggieWX, 2021-04-16:
Hi,

Could i find out how do you use the move_group_interface.py. I am using the locobot robot model.

Jekyll1021, 2021-04-19:
http://docs.ros.org/en/indigo/api/moveit_tutorials/html/doc/pr2_tutorials/planning/scripts/doc/move_group_python_interface_tutorial.html

checkout this tutorial. PyRobot only provide a bridge to moveit!, the API and functionalities are left as-is.

closed 2021-04-19 by Jekyll1021