facebookresearch/pyrobot #153: get_current_pcd from camera.py returns empty list (again)

Jendker, 2021-05-03:
I am running into the same issue with `get_current_pcd` as before in another issue: https://github.com/facebookresearch/pyrobot/issues/139 after I have updated Develop branch to the latest version. Were there any changes to `get_current_pcd`?

## Steps to reproduce

  1. install pyrobot with either python2 or python3 on locobot
  2. run `roslaunch locobot_control main.launch use_arm:=true use_camera:=true use_base:=true use_rviz:=false`
  3. `get_current_pcd` function returns empty list if executed in script (script below)

## Observed Results

  * I have reinstalled the robot because of problem with the software and now the function `get_current_pcd` (https://github.com/facebookresearch/pyrobot/blob/master/src/pyrobot/locobot/camera.py#L68) returns empty array. rostopic `/camera/depth_registered/points` publishes the pointcloud fine, though. Verified with Rviz.

## Expected Results

  * Before the software reinstall the pointcloud was returned correctly.

## Relevant Code

```
from pyrobot import Robot

bot = Robot('locobot')
print('pcl', bot.camera.get_current_pcd(True))  # or False, does not matter
```

Jekyll1021, 2021-05-04:
Hi @Jendker could you print `depth` at https://github.com/facebookresearch/pyrobot/blob/4b3f0135f2610646e89840329cd1a2db3db5ee63/src/pyrobot/locobot/camera.py#L446?

Jendker, 2021-05-05:
Thank you for you answer!
While I am still able to see the pointcloud in RViz on `/camera/depth_registered/points` topic, `depth` is all zeros. If that matters, we have not updated any packages on the robot in the meantime.

Jekyll1021, 2021-05-05:
Could you confirm the entire depth array is zero (unlike #139 where you get some numerically small number that get filtered out downstream)? If so, is there any error / warning in the roslaunch window?

Jendker, 2021-05-05:
I can confirm that entire `depth` array is zero and there are no errors in the roslaunch window:

```
roslaunch locobot_control main.launch use_arm:=true use_camera:=true use_base:=true use_rviz:=false
... logging to /home/locobot/.ros/log/3c4860e8-adc9-11eb-900b-48f17fded622/roslaunch-locobot-19738.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://locobot:39827/

SUMMARY
========

PARAMETERS
 * /bumper2pointcloud/pointcloud_radius: 0.24
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -1.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
[... 601 lines cut ...]
Loading 'move_group/MoveGroupQueryPlannersService'...
Loading 'move_group/MoveGroupStateValidationService'...
[ INFO] [1620236552.981960066]: 

********************************************************
* MoveGroup using: 
*     - ApplyPlanningSceneService
*     - ClearOctomapService
*     - CartesianPathService
*     - ExecuteTrajectoryAction
*     - GetPlanningSceneService
*     - KinematicsService
*     - MoveAction
*     - PickPlaceAction
*     - MotionPlanService
*     - QueryPlannersService
*     - StateValidationService
********************************************************

[ INFO] [1620236552.982034774]: MoveGroup context using planning plugin ompl_interface/OMPLPlanner
[ INFO] [1620236552.982061540]: MoveGroup context initialization complete

You can start planning now!
```
I can try reinstalling the whole robot, maybe that will help. Should I use master or Develop branch on the new installation?

Jekyll1021, 2021-05-05:
I tested on master branch. Could you check `get_rgb_depth()` is returning value properly?

Jendker, 2021-05-05:
I'm on master branch as well now. The first component returns has values, only depth is all zeros. Calling `rostopic echo /camera/depth_registered/points` still gives me values != 0 as expected.

Jekyll1021, 2021-05-05:
Could you try `rostopic echo /camera/aligned_depth_to_color/image_raw`?

Jendker, 2021-05-05:
Still looks good!

Jekyll1021, 2021-05-05:
Could you poke into this line https://github.com/facebookresearch/pyrobot/blob/13a70d2427417756504ca4ff3cf99a33f15abbde/src/pyrobot/core.py#L364
to see if self.depth_img update makes sense? Reinstalling would most likely help, but I hope we can pinpoint the problem before doing so - if something pops up after reinstall, it would be exponentially harder to debug.

Jendker, 2021-05-05:
`self.depth_img` looks fine, I am getting non-zero outputs

Jekyll1021, 2021-05-05:
Okay sounds good. At this point, I actually suggest against reinstalling, since the problem should be somewhere between subscribe image and returning depth. 

Could you check the `depth` here
https://github.com/facebookresearch/pyrobot/blob/13a70d2427417756504ca4ff3cf99a33f15abbde/src/pyrobot/core.py#L411

by calling `get_rgb_depth`?

Jendker, 2021-05-05:
Hmm looks like python2 and 3 compatibility issue. I am still running python2 and the function is running int division by  `self.configs.CAMERA.DEPTH_MAP_FACTOR`. I have fixed that by adding float(...) here https://github.com/facebookresearch/pyrobot/blob/13a70d2427417756504ca4ff3cf99a33f15abbde/src/pyrobot/core.py#L397 and here https://github.com/facebookresearch/pyrobot/blob/13a70d2427417756504ca4ff3cf99a33f15abbde/src/pyrobot/core.py#L412 but I am still getting empty array. What other place do I need to adjust?

Jekyll1021, 2021-05-05:
I want to see if the raw depth before division is giving values (and also in meter). If the raw depth is giving actual value but zeroed out after a division, realsense might be giving meter reading instead of millimeter reading, or there's some other problem, like the formatting problem you just caught.

https://github.com/facebookresearch/pyrobot/blob/13a70d2427417756504ca4ff3cf99a33f15abbde/src/pyrobot/core.py#L411

Jendker, 2021-05-06:
I see, so I may need to adjust the units myself.
Here are the values of `depth` I am getting in the line mentioned by you:

```
[[864 875 875 ... 921 921 921]
 [861 871 871 ... 921 921 921]
 [861 871 871 ... 917 917 918]
 ...
 [526 526 526 ... 521 521 521]
 [526 526 526 ... 521 521 521]
 [526 526 525 ... 520 520 520]]
```

Jekyll1021, 2021-05-06:
> I see, so I may need to adjust the units myself.
> Here are the values of `depth` I am getting in the line mentioned by you:
>
> ```
> [[864 875 875 ... 921 921 921]
>  [861 871 871 ... 921 921 921]
>  [861 [...]

These values look like proper millimeter readings to me. Does `depth` give all zeros at https://github.com/facebookresearch/pyrobot/blob/13a70d2427417756504ca4ff3cf99a33f15abbde/src/pyrobot/core.py#L412?

If so, could you check what's the type and value of `depth` and `self.configs.CAMERA.DEPTH_MAP_FACTOR`?

Jendker, 2021-05-06:
`depth` is all zeros after division in line 412 and is a np.ndarray with dtype uint16. `self.configs.CAMERA.DEPTH_MAP_FACTOR` is of int type.

Jekyll1021, 2021-05-06:
This is indeed the problem. Could you change the line to `depth = depth.astype(np.float32) / 1000` ?

Jendker, 2021-05-06:
Hmm after the adjustment `depth` value is:

```
[[0.864 0.875 0.875 ... 0.921 0.921 0.921]
 [0.861 0.871 0.871 ... 0.921 0.921 0.921]
 [0.861 0.871 0.871 ... 0.917 0.917 0.918]
 ...
 [0.526 0.526 0.526 ... 0.521 0.521 0.521]
 [0.526 0.526 0.526 ... 0.521 0.521 0.521]
 [0.525 0.525 0.525 ... 0.52  0.52  0.52 ]]
```

but the list returned by `get_current_pcd` is still empty. Do you recall if there is any other division I should to adjust?

Jekyll1021, 2021-05-06:
Could you check `depth` before and after this line?

https://github.com/facebookresearch/pyrobot/blob/13a70d2427417756504ca4ff3cf99a33f15abbde/src/pyrobot/locobot/camera.py#L441

Jendker, 2021-05-06:
No problem!

Before:
```
[[0.864 0.874 0.874 ... 0.921 0.921 0.921]
 [0.861 0.871 0.871 ... 0.921 0.921 0.921]
 [0.861 0.871 0.871 ... 0.918 0.918 0.918]
 ...
 [0.527 0.527 0.527 ... 0.522 0.521 0.521]
 [0.527 0.527 0.527 ... 0.522 0.521 0.521]
 [0.526 0.526 0.526 ... 0.521 0.52  0.52 ]]
```
after:
```
[0.000864 0.000874 0.000874 ... 0.000521 0.00052  0.00052 ]
```

Jekyll1021, 2021-05-06:
That's why. Change the line to `depth = depth_im.reshape(-1)` should solve the problem!

Jendker, 2021-05-06:
We are back in business! The second returned value of `get_current_pcd` is of np.ndarray and has dtype uint8 in contrast to depth which is float32. Is that fine?

Jekyll1021, 2021-05-06:
The second return value is RGB of point cloud which is suppose to be uint8.

closed 2021-05-06 by Jekyll1021

Jendker, 2021-05-06:
Thank you for the help!