facebookresearch/pyrobot #158: Installer potentially out of date with current codebase

tkim61, 2021-05-25:
## Steps to reproduce

  1. Tried to use functions from pyrobot/utils/util for the active camera demo
  2. Found that the directory pyrobot/utils did not exist - I had no access to the robot for more than a year, last used march 2019 to best of my knowledge 
  3. Tried to use the "install pyrobot only" instructions to get an updated codebase from github
  4. This made the running ROS break - error message said it was looking for a file `
pyrobot/robots/LoCoBot/locobot_description/urdf/locobot_description.urdf`, which does not exist and is named only in that directory's .gitignore 
 5. I tried deleting all files and doing a fresh install thinking maybe I needed to update something that wasn't in the pyrobot section 
 6. After updating most of my software, the installer finished without crashing, but did not install the camera configuration files, telling me to install them manually
 7. However, running ROS at this time still doesn't work, and now crashes

## Observed Results
this is the output of the most recent crash message, not sure where the log file is in the filesystem
```
[main.launch] is neither a launch file in package [locobot_control] nor is [locobot_control] a launch file name
The traceback for the exception was written to the log file
```
## Expected Results
A clean, functional and up to date install of pyrobot, the usage of which agrees with the tutorial.
  
## Relevant Code
-unsure, I suspect `locobot_install_all.sh` is out of sync with the rest of the codebase, writing files that are supposed to work with some older versions

BlackDChase, 2021-11-06:
Any fix for this found? @tkim61

srama2512, 2022-06-30:
@tkim61  @BlackDChase - Any luck with this? I'm also running into the same issue.